ccMSC/ckb #311: Sabre RGB not detected by ckb

DivineComedy11, 2016-04-25:
**Before I go into the issue, I'd like to thank ccMSC for their tireless efforts with this driver!**

I have a red K70 (which is detected fine), and have recently purchased an RGB Sabre, which ckb unfortunately cannot see. 

I have a feeling this is because I have a slightly newer model than the one previously supported, or at least one with a different bus ID as I get the following output when I run *lsusb -d 1b1c*:
```
Bus 003 Device 003: ID 1b1c:1b2f Corsair  
```  
(Alongside the 1b09 ID for the K70)

Stopping the daemon and running it again just picks up the k70, as you might expect. I've tried digging around, but I get the impression it's slightly more involved than adding a string somewhere, so any input would be much appreciated.

PLUT094, 2016-04-25:
I have the same issue as well. Just bought Sabre and it does not detect it.

ccMSC, 2016-04-26:
Actually fairly simple :) Added here: https://github.com/ccMSC/ckb/tree/sabre

Let me know if it's working properly or if there's something else I need to do.

DivineComedy11, 2016-04-26:
Hi ccMSC, I gave it a try and the mouse is detected by the application, but for some reason is now unable to move! When I ran the ckb-daemon I got this __(Interrupt is mine)__:
```

 ckb: Corsair RGB driver beta-v0.2.5+s01
[I] Root controller ready at /dev/input/ckb0
[I] Connecting Corsair K70R Gaming Keyboard at /dev/input/ckb2
[I] Connecting Corsair Gaming Sabre RGB Mouse at /dev/input/ckb1
[I] Starting input thread for /dev/input/ckb1
[I] Starting input thread for /dev/input/ckb2
[I] Setup finished for /dev/input/ckb2
[E] os_sendindicators (usb_linux.c:84): Connection timed out
[E] os_usbsend (via firmware.c:15): Connection timed out
...
[E] _usbrecv (via firmware.c:15): Too many send/recv failures. Dropping.
[W] _start_dev (device.c:24): Unable to load firmware version/poll rate
[I] Setup finished for /dev/input/ckb1
[E] os_sendindicators (usb_linux.c:84): Connection timed out
[E] os_sendindicators (usb_linux.c:84): Connection timed out
[E] os_usbsend (via device_keyboard.c:38): No such file or directory
[I] Attempting reset...
[E] os_usbsend (via device_keyboard.c:38): No such file or directory
...
[E] os_resetusb (via usb.c:169): resetusb failed: No such device
[E] usb_tryreset (usb.c:177): Reset failed. Disconnecting.
[I] Disconnecting /dev/input/ckb1
[I] Stopping input thread for /dev/input/ckb1
[I] Removed device path /dev/input/ckb1
[I] Connecting Corsair Gaming Sabre RGB Mouse at /dev/input/ckb1
[I] Starting input thread for /dev/input/ckb1
[E] os_usbrecv (via led_keyboard.c:207): Connection timed out
[W] _start_dev (device.c:45): Unable to load hardware profile
[I] Setup finished for /dev/input/ckb1
[E] os_usbsend (via device_keyboard.c:38): No such file or directory
[I] Attempting reset...
[E] os_usbsend (via device_keyboard.c:38): No such file or directory
...
^C
[I] Caught signal 2
[E] os_usbsend (via device_keyboard.c:38): No such file or directory
[E] usb_tryreset (usb.c:177): Reset failed. Disconnecting.
[I] Disconnecting /dev/input/ckb1
[I] Removed device path /dev/input/ckb1
[I] Disconnecting /dev/input/ckb2
[I] Removed device path /dev/input/ckb2
[I] Closing root controller
[I] Removed device path /dev/input/ckb0

```
I noticed that the application polls the mouse and prints c:38 until it gives up and drops the Sabre as a menu option. Keyboard also works fine throughout process.

Thanks for your help.

PLUT094, 2016-04-29:
@ccMSC Any update to this? I've had same issues as @DivineComedy11 mouse does not move but it does detect it..

Thanks!
-Mike

ccMSC, 2016-05-10:
Sorry, I've been away from the project for the past week so I didn't have a chance to look into it.

I thought it might have been caused by the firmware, so I've merged the `fw2kb` branch. Might solve the problem. If not, could you provide me the full USB information for the device, via

`lsusb -vd 1b1c:1b2f`

DivineComedy11, 2016-05-10:
Gave it a try, no luck unfortunately. 

Here's the output:

```
Bus 003 Device 002: ID 1b1c:1b2f Corsair 
Couldn't open device, some information will be missing
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               2.00
  bDeviceClass            0 
  bDeviceSubClass         0 
  bDeviceProtocol         0 
  bMaxPacketSize0        64
  idVendor           0x1b1c Corsair
  idProduct          0x1b2f 
  bcdDevice            2.03
  iManufacturer           1 
  iProduct                2 
  iSerial                 3 
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength           84
    bNumInterfaces          3
    bConfigurationValue     1
    iConfiguration          0 
    bmAttributes         0xa0
      (Bus Powered)
      Remote Wakeup
    MaxPower              300mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      1 Boot Interface Subclass
      bInterfaceProtocol      2 Mouse
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.11
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength      58
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x81  EP 1 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0008  1x 8 bytes
        bInterval               8
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        1
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      0 
      bInterfaceProtocol      0 
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.11
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength     145
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x82  EP 2 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0040  1x 64 bytes
        bInterval               1
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        2
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      0 
      bInterfaceProtocol      0 
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.11
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength      25
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x03  EP 3 OUT
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0040  1x 64 bytes
        bInterval               1
```

Thank you for your help.

orangeturtle739, 2016-06-02:
I just had the same problem on the sabre branch. It seems that the issue was in `src/ckb-daemon/usb.h` on line 66. For the `IS_MOUSE` check, it did not check to see if the mouse was a new sabre mouse. I forked the repository and fixed it: [https://github.com/orangeturtle739/ckb/tree/sabre](https://github.com/orangeturtle739/ckb/tree/sabre), and now it works.

PLUT094, 2016-06-02:
@orangeturtle739 It works!!! Thank you

orangeturtle739, 2016-06-02:
I opened a pull request for the fix: [https://github.com/ccMSC/ckb/pull/367](https://github.com/ccMSC/ckb/pull/367).

DivineComedy11, 2016-06-03:
Good spot, nice one!

PLUT094, 2016-06-03:
One issue I notice, not sure if anyone else has similar problems. When I turn on my computer in the morning the mouse doesn't move/detect it and I would have to either replug it back in OR run quickinstall again to get it to work.

Any ideas? @orangeturtle739

orangeturtle739, 2016-06-03:
Not quite sure. I just tried restarting my computer, and the mouse worked fine. I am running Kubuntu 16.04.

PLUT094, 2016-06-03:
maybe a mac issue..

orangeturtle739, 2016-06-03:
I can't get it to work on my mac at all. What version of OS X are you running?

tatokis, 2016-06-03:
There's another bug report, can't remember which one, but there was a 
fix about inconsistent Mac behaviour. Maybe the fork doesn't have that 
fix included?

orangeturtle739, 2016-06-03:
I just forked it yesterday, and put the fix on the sabre branch, which had the fw2kb branch already merged into it. So unless this mac fix was on the testing branch, I should have gotten it.

tatokis, 2016-06-03:
https://github.com/ccMSC/ckb/issues/233#issuecomment-219214176

Maybe the fix wasn't pushed to source yet

PLUT094, 2016-06-03:
@orangeturtle739 I'm running 10.11.4

orangeturtle739, 2016-06-03:
It looks like it was fixed (https://github.com/ccMSC/ckb/issues/233#issuecomment-219214176, with this package: https://dl.dropboxusercontent.com/u/16424004/ckb/ckb%20v0.2.5%2Bf05%2Blogging.pkg) but the source was not pushed, so we can't combine it with this fix.

ccMSC, 2016-06-18:
Hey everyone, sorry for my absence but I finally got a chance to look at this again. I'm moving all new devices into the `newdev` branch, which now has the latest source changes as well as the IS_MOUSE fix. Let me know if it's working for you.

orangeturtle739, 2016-06-20:
Thanks! I will try it out on my mac soon.

DrDeath13, 2016-07-25:
Changing the hwid worked with my Sabre. Probably another new version. Mine has 
`Bus 008 Device 002: ID 1b1c:1b32 Corsair `
I replaced the 1b14 @ usb.h and it got recognized by ckb like a charm. Don't know how to get it upstream so if @ccMSC or someone else can it'll be great for future updates. At first I tried to add the model as SABRE_ON (as Optical New) but I got some errors building it so, replacing hwid seemed easier than debugging. 
And since I'm here, one request : is it possible to change  the percentage of lightning seperately on each device? If I change it on Sabre it changes the keyboards (Strafe) as well and vice versa. 
Thanks for your great work

entisocial, 2016-10-10:
Hi, I'm new to all of this (since I got the mouse yesterday). I've got a sabre RGB with ID 1b1c:1b2f and I'm running macOS 10.12. I've tried the default beta 2.0.6 > my mouse isn't recognized. I've tried the fork from https://github.com/orangeturtle739/ckb/tree/sabre > still not recognized. I've tried the newdev branch and even though my device is present in the usb.h my mouse is still not recognized. Am I missing something am I supposed to do something else than just compile from the newdev branch?

ollieatkinson, 2017-02-08:
I have this mouse:

```
$ system_profiler SPUSBDataType
...
        Corsair Gaming Sabre RGB Mouse:

          Product ID: 0x1b2f
          Vendor ID: 0x1b1c  (CORSAIR MEMORY INC.)
          Version: 2.03
          Serial Number: 0D00B015AEBD1840562C571FF5001946
          Speed: Up to 12 Mb/sec
          Manufacturer: Corsair
          Location ID: 0x14700000 / 8
          Current Available (mA): 500
          Current Required (mA): 300
          Extra Operating Current (mA): 0
...
```

And I can see it defined in the `usb.h` file

```
#define P_SABRE_N       0x1b2f  /* new? */
#define P_SABRE_N_STR   "1b2f"
#define IS_SABRE(kb)    ((kb)->vendor == V_CORSAIR && ((kb)->product == P_SABRE_O || (kb)->product == P_SABRE_L || (kb)->product == P_SABRE_N))
```

But I keep getting `No devices connected` - any recommendations?

orangeturtle739, 2017-02-09:
I never got it to work on a mac, and now I don't have a mac anymore to test on.

Perhaps try [ckb-next](https://github.com/mattanger/ckb-next)? This repository seems to be no longer maintained, and they might have fixed it there.

ollieatkinson, 2017-02-09:
@orangeturtle739 That worked - thanks for the tip.

ekcg, 2017-06-20:
Hi, I recently bought a Sabre (CH-9303011-NA). 
I first installed the binary file, ckb beta-v0.2.6
it says no devices connected.

Then, I tried https://github.com/mattanger/ckb-next#upgrading-source
it says "No devices connected. Warning: Driver version mismatch."

How do I find a matching driver? I believe its beta-v0.2.8, and how do I install that driver??

ollieatkinson, 2017-06-20:
Ensure you are using https://github.com/mattanger/ckb-next and the daemon is running @ekcg

ekcg, 2017-06-20:
How do I ensure daemon is running?? @ollieatkinson

ekcg, 2017-06-21:
It worked! I'm guessing the older versions wasn't completely uninstalled, so the error keeps popping up. I uninstalled Xcode, qt, and deleted all the folders of the old ckb's. installed the binary for 0.2.8 and worked like a charm!