ccMSC/ckb #313: K95 RGB repeatedly disconnecting when running ckb

olofson, 2016-04-26:
* K95 RGB (New version of the old Corsair logo.)
* Lenovo X230
* Kubuntu (Linux 3.13.0-85-generic # 129-Ubuntu SMP Thu Mar 17 20:50:15 UTC 2016 x86_64 x86_64 x86_64 GNU/Linux)
* ckb beta-v0.2.4 and beta-v0.2.5
* usbhid.quirks=0x1B1C:0x1B11:0x20000408 (without this, the system gets stuck when booting with the keyboard connected, and reconnecting after disconnect takes "forever," and eventually makes the kernel driver fail, requiring a reboot)

I get a USB disconnect every few minutes. Doesn't matter which USB port(s) I use.

If I quit ckb, the disconnects stop, and the keyboard works reliably.
```
[ 9713.994789] usb 3-1: USB disconnect, device number 53
[ 9714.320044] usb 3-1: new full-speed USB device number 54 using xhci_hcd
[ 9714.337879] usb 3-1: New USB device found, idVendor=1b1c, idProduct=1b11
[ 9714.337888] usb 3-1: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[ 9714.337892] usb 3-1: Product: Corsair K95 RGB Gaming Keyboard
[ 9714.337895] usb 3-1: Manufacturer: Corsair
[ 9714.337898] usb 3-1: SerialNumber: 0A00E00DAEA7A44454E0C100F5001945
[ 9714.339729] input: Corsair Corsair K95 RGB Gaming Keyboard  as /devices/pci0000:00/0000:00:14.0/usb3/3-1/3-1:1.0/input/input216
[ 9714.340119] hid-generic 0003:1B1C:1B11.00C8: input,hidraw0: USB HID v1.11 Keyboard [Corsair Corsair K95 RGB Gaming Keyboard ] on usb-0000:00:14.0-1/input0
[ 9714.341409] input: Corsair Corsair K95 RGB Gaming Keyboard  as /devices/pci0000:00/0000:00:14.0/usb3/3-1/3-1:1.1/input/input217
[ 9714.341668] hid-generic 0003:1B1C:1B11.00C9: input,hidraw1: USB HID v1.11 Keyboard [Corsair Corsair K95 RGB Gaming Keyboard ] on usb-0000:00:14.0-1/input1
[ 9714.342559] hid-generic 0003:1B1C:1B11.00CA: hiddev0,hidraw2: USB HID v1.11 Device [Corsair Corsair K95 RGB Gaming Keyboard ] on usb-0000:00:14.0-1/input2
[ 9714.343545] hid-generic 0003:1B1C:1B11.00CB: hiddev0,hidraw3: USB HID v1.11 Device [Corsair Corsair K95 RGB Gaming Keyboard ] on usb-0000:00:14.0-1/input3
[ 9714.624693] input: ckb1: Corsair K95 RGB Gaming Keyboard as /devices/virtual/input/input218
[ 9714.625067] input: ckb1: Corsair K95 RGB Gaming Keyboard as /devices/virtual/input/input219
```

I also have another system, which works perfectly fine. Slightly older (tribal logo) K95 RGB running Ubuntu Studio with a 3.13.0-85-lowlatency kernel and ckb beta-v0.2.4+t02.

Things I consider trying:
* Swapping the two keyboards.
* Trying beta-v0.2.5 on the working system currently running beta-v0.2.4+t02.

Any other ideas?

olofson, 2016-04-26:
**Update:** Have been using the "offending" keyboard on by Ubuntu Studio system with ckb beta-v0.2.4+t02 for over an hour now, without any issues.

olofson, 2016-04-27:
To clarify, the disconnect issues I'm having seem to be tied to the Lenovo X230 (as opposed to keyboard or ckb version), but only occur when using ckb.

Is there anything ckb does that is known to cause problems with some USB root hubs? I haven't managed to find anything relevant related to the X230, so I'm out of ideas here.

ccMSC, 2016-04-28:
Assuming it's not a power issue (you could probably verify this with a powered hub), only other thing I can think of is the volume of USB traffic caused by sending animations. You might be able to work around it by decreasing your frame rate under `Settings > More Settings > FPS`.

olofson, 2016-04-28:
There is animation traffic going on even when not using any animations...? And does that increase the power draw significantly? (I'm not using animations, but quitting ckb eliminates the disconnect problem.)

Also, disconnects happen with any USB ports I've tried so far, including the USB3 ones.

I'll see if I can figure out some more things to try...

olofson, 2016-04-28:
Tried a powered USB2 hub, and lowering the animation frame rate to 5 Hz. Still having disconnects every few minutes when ckb is running. Quit ckb, and the problem disappears. (But of course, G keys etc aren't working without it, so that's not really an option for me.)

olofson, 2016-05-02:
I captured the traffic during one of these disconnect/reconnect cycles using usbmon:
[k95-rgb-usbmon.txt](https://github.com/ccMSC/ckb/files/245204/k95-rgb-usbmon.txt)

olofson, 2016-05-12:
With fw2kb, I see the same problem, but there may be an additional hint; "USBDEVFS_CONTROL failed cmd" etc:
```
[241218.898852] usb 3-1.3: usbfs: USBDEVFS_CONTROL failed cmd ckb-daemon rqt 161 rq 1 len 64 ret -110
[241251.933972] usb 3-1.3: USB disconnect, device number 100
[241252.157891] usb 3-1.3: new full-speed USB device number 101 using xhci_hcd
[241252.176222] usb 3-1.3: New USB device found, idVendor=1b1c, idProduct=1b11
[241252.176229] usb 3-1.3: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[241252.176233] usb 3-1.3: Product: Corsair K95 RGB Gaming Keyboard 
[241252.176236] usb 3-1.3: Manufacturer: Corsair
[241252.176239] usb 3-1.3: SerialNumber: 0A00E00DAEA7A44454E0C100F5001945
[241252.177985] input: Corsair Corsair K95 RGB Gaming Keyboard  as /devices/pci0000:00/0000:00:14.0/usb3/3-1/3-1.3/3-1.3:1.0/input/input242
[241252.178696] hid-generic 0003:1B1C:1B11.00DE: input,hidraw3: USB HID v1.11 Keyboard [Corsair Corsair K95 RGB Gaming Keyboard ] on usb-0000:00:14.0-1.3/input0
[241252.180159] input: Corsair Corsair K95 RGB Gaming Keyboard  as /devices/pci0000:00/0000:00:14.0/usb3/3-1/3-1.3/3-1.3:1.1/input/input243
[241252.180775] hid-generic 0003:1B1C:1B11.00DF: input,hidraw4: USB HID v1.11 Keyboard [Corsair Corsair K95 RGB Gaming Keyboard ] on usb-0000:00:14.0-1.3/input1
[241252.182157] hid-generic 0003:1B1C:1B11.00E0: hiddev0,hidraw5: USB HID v1.11 Device [Corsair Corsair K95 RGB Gaming Keyboard ] on usb-0000:00:14.0-1.3/input2
[241252.183281] hid-generic 0003:1B1C:1B11.00E1: hiddev0,hidraw6: USB HID v1.11 Device [Corsair Corsair K95 RGB Gaming Keyboard ] on usb-0000:00:14.0-1.3/input3
[241252.430003] input: ckb1: Corsair K95 RGB Gaming Keyboard as /devices/virtual/input/input244
[241252.430847] input: ckb1: Corsair K95 RGB Gaming Keyboard as /devices/virtual/input/input245
[241259.213828] usb 3-1.3: usbfs: USBDEVFS_CONTROL failed cmd ckb-daemon rqt 161 rq 1 len 64 ret -110
[241315.950834] usb 3-1.3: USB disconnect, device number 101
[241316.167229] usb 3-1.3: new full-speed USB device number 102 using xhci_hcd
[241316.185450] usb 3-1.3: New USB device found, idVendor=1b1c, idProduct=1b11
[241316.185458] usb 3-1.3: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[241316.185464] usb 3-1.3: Product: Corsair K95 RGB Gaming Keyboard 
[241316.185468] usb 3-1.3: Manufacturer: Corsair
[241316.185472] usb 3-1.3: SerialNumber: 0A00E00DAEA7A44454E0C100F5001945
[241316.187729] input: Corsair Corsair K95 RGB Gaming Keyboard  as /devices/pci0000:00/0000:00:14.0/usb3/3-1/3-1.3/3-1.3:1.0/input/input246
[241316.188067] hid-generic 0003:1B1C:1B11.00E2: input,hidraw3: USB HID v1.11 Keyboard [Corsair Corsair K95 RGB Gaming Keyboard ] on usb-0000:00:14.0-1.3/input0
[241316.189503] input: Corsair Corsair K95 RGB Gaming Keyboard  as /devices/pci0000:00/0000:00:14.0/usb3/3-1/3-1.3/3-1.3:1.1/input/input247
[241316.190415] hid-generic 0003:1B1C:1B11.00E3: input,hidraw4: USB HID v1.11 Keyboard [Corsair Corsair K95 RGB Gaming Keyboard ] on usb-0000:00:14.0-1.3/input1
[241316.191878] hid-generic 0003:1B1C:1B11.00E4: hiddev0,hidraw5: USB HID v1.11 Device [Corsair Corsair K95 RGB Gaming Keyboard ] on usb-0000:00:14.0-1.3/input2
[241316.192939] hid-generic 0003:1B1C:1B11.00E5: hiddev0,hidraw6: USB HID v1.11 Device [Corsair Corsair K95 RGB Gaming Keyboard ] on usb-0000:00:14.0-1.3/input3
[241316.442807] input: ckb1: Corsair K95 RGB Gaming Keyboard as /devices/virtual/input/input248
[241316.443926] input: ckb1: Corsair K95 RGB Gaming Keyboard as /devices/virtual/input/input249
[241323.559303] usb 3-1.3: usbfs: USBDEVFS_CONTROL failed cmd ckb-daemon rqt 161 rq 1 len 64 ret -110
```
(Might have a go with firmware 2.04 later, and see if that fixes my disconnect issue. 2.04 is why I'm trying fw2kb in the first place.)

olofson, 2016-06-07:
Update: Same problem with firmware 2.04; the keyboard still disconnects every few seconds when using ckb on this Lenovo X230.

Also, beta 0.2.5 (which works just fine on my Ubuntu Studio PC at home) doesn't work at all with fw 2.04. This happens:
```
[  158.699800] input: Corsair Corsair K95 RGB Gaming Keyboard  as /devices/pci0000:00/0000:00:14.0/usb3/3-1/3-1.1/3-1.1.3/3-1.1.3:1.0/input/input28
[  158.700708] hid-generic 0003:1B1C:1B11.000A: input,hidraw3: USB HID v1.11 Keyboard [Corsair Corsair K95 RGB Gaming Keyboard ] on usb-0000:00:14.0-1.1.3/input0
[  158.701800] input: Corsair Corsair K95 RGB Gaming Keyboard  as /devices/pci0000:00/0000:00:14.0/usb3/3-1/3-1.1/3-1.1.3/3-1.1.3:1.1/input/input29
[  158.701960] hid-generic 0003:1B1C:1B11.000B: input,hiddev0,hidraw4: USB HID v1.11 Keyboard [Corsair Corsair K95 RGB Gaming Keyboard ] on usb-0000:00:14.0-1.1.3/input1
[  158.701984] usb 3-1.1.3: usbfs: process 1328 (ckb-daemon) did not claim interface 0 before use
[  158.701990] usb 3-1.1.3: usbfs: process 1328 (ckb-daemon) did not claim interface 1 before use
[  159.092137] usb 3-1.1.3: reset full-speed USB device number 8 using xhci_hcd
[  159.109098] xhci_hcd 0000:00:14.0: xHCI xhci_drop_endpoint called with disabled ep ffff8800363fd6c0
[  159.109106] xhci_hcd 0000:00:14.0: xHCI xhci_drop_endpoint called with disabled ep ffff8800363fd680
[  159.109110] xhci_hcd 0000:00:14.0: xHCI xhci_drop_endpoint called with disabled ep ffff8800363fd640
[  159.392204] usb 3-1.1.3: reset full-speed USB device number 8 using xhci_hcd
[  159.409177] xhci_hcd 0000:00:14.0: xHCI xhci_drop_endpoint called with disabled ep ffff8800363fd6c0
[  159.409185] xhci_hcd 0000:00:14.0: xHCI xhci_drop_endpoint called with disabled ep ffff8800363fd680
[  159.409189] xhci_hcd 0000:00:14.0: xHCI xhci_drop_endpoint called with disabled ep ffff8800363fd640
[  159.692216] usb 3-1.1.3: reset full-speed USB device number 8 using xhci_hcd
[  159.709062] xhci_hcd 0000:00:14.0: xHCI xhci_drop_endpoint called with disabled ep ffff8800363fd6c0
[  159.709068] xhci_hcd 0000:00:14.0: xHCI xhci_drop_endpoint called with disabled ep ffff8800363fd680
[  159.709072] xhci_hcd 0000:00:14.0: xHCI xhci_drop_endpoint called with disabled ep ffff8800363fd640
[  159.992375] usb 3-1.1.3: reset full-speed USB device number 8 using xhci_hcd
[  160.009440] xhci_hcd 0000:00:14.0: xHCI xhci_drop_endpoint called with disabled ep ffff8800363fd6c0
[  160.009450] xhci_hcd 0000:00:14.0: xHCI xhci_drop_endpoint called with disabled ep ffff8800363fd680
[  160.009454] xhci_hcd 0000:00:14.0: xHCI xhci_drop_endpoint called with disabled ep ffff8800363fd640
[  160.292369] usb 3-1.1.3: reset full-speed USB device number 8 using xhci_hcd
[...]
```
The only way I can bring the keyboard back to life at all after that is to reboot the system. Stopping ckb and unplugging and replugging the keyboard doesn't help. It's gone, and the keyboard fw switches to demo mode.

(Since beta 0.2.5 works perfectly fine with fw 2.04 on the other machine, I'm guessing this might be related to the disconnect issue I'm having with this machine, and not a general 2.0.5/2.04 issue.)

closed 2016-06-13 by ccMSC