ccMSC/ckb #314: Mac OS X : Does not work anymore with 2.04 firmware

4t0m1k, 2016-04-27:
Hello,

I use my K65 RGB keyboard on PC and Mac (with a KVM switch) and all was working perfectly but since I've updated the firmware to 2.04 yesterday, my keyboard is no more recognized on Mac. I've updated ckb to 0.25 (I was on 0.24) but no more chance.

Is there a way to fix this or to roll back to previous firmware version ?

Thank you very much for your work, without ckb, I wouldn't buy this keyboard. Corsair company should work with you !

joinscome, 2016-04-27:
Hi

Same problem here, but with my mouse (M65) which has been udated to the latest firmware a few days ago. No problem with my K95 keyboard, but I didn't install the latest update yet.

As I wrote in another topic, I have to use the version 0.2.4 of CKB. With the 0.2.5, neither my keyboard, nor my mouse are recognized, even before the firmware update of my M65. 

But thank you for your great work ! I'm sure this will be fixed soon ;)

Mills00013, 2016-04-28:
This is probably the same issue as #298 

It's being worked on.

ccMSC, 2016-04-28:
Closing this issue as duplicate - feel free to continue discussion in the other topic. I think it's close to being solved now; if you need an OS X binary I should be able to provide one soon.

closed 2016-04-28 by ccMSC

Handsomfreddy, 2016-05-03:
Great thanks so much it is much appreciated!!! : )