ccMSC/ckb #316: K95 quits sending keys when ckb starts

phorgan1, 2016-04-28:
On boot, my new K95 works fine, but when ckb starts (I had it start on login). The keys stopped being sent to X. The mouse worked fine so I could reboot, and again keys worked fine until ckb showed up in the taskbar and at that instant the keys stopped working again. On the keyboard I could do things like change the brightness of the lights, but it quit sending.

Linux localhost.localdomain 4.4.6-201.fc22.x86_64 #1 SMP Wed Mar 30 18:30:16 UTC 2016 x86_64 x86_64 x86_64 GNU/Linux

This happened with both the master and the testing branches. I wasn't able to test whether the keyboard would start working if I killed ckb because I couldn't think of a way off hand to do it with just a mouse. I'm a command line guy.

ckb was in the taskbar and it would run just fine, but sending animations seemed to do nothing.  I unclicked that I wanted ckb to start at login and rebooted and the keyboard works fine now.

I am a retired system programmer with experience mostly in C/C++ so if there is anything I can do to track this down, I'm pretty motivated.

Rebooting

tatokis, 2016-04-28:
It's probably this issue.
For now you should either downgrade the firmware or make the 
application not run automatically, until this is fixed.

https://github.com/ccMSC/ckb/issues/298

ccMSC, 2016-04-28:
don't think the K95 ever got the new firmware? Unless it was moved to a different URL...

To clarify, can you try to find out whether it's the GUI process (`ckb`, runs as a normal user) or the driver (`ckb-daemon`, runs as root) which is causing the issue? For instance, if you close the GUI, does it go back to normal, or is it necessary to shutdown the daemon as well?

tatokis, 2016-04-28:
Apologies, my mistake, I completely missed out the "K95" part.

Can you also try running this in a terminal and then pressing some keys?
```
cat /dev/input/ckb1/notify1
```

phorgan1, 2016-04-28:
The daemon is already running and the keyboard is fine, then I run ckb and instantly no joy. I plug in another keyboard and kill ckb and the K95 still doesn't work. I unplug and replug the USB cables from the K95 and it starts working. Then if I run ckb it doesn't affect the K95, but ckb says that it doesn't find any keyboards.

Whether or not the keyboard is hung, cat /dev/input/ckb1/notify1 just says no such file.

ccMSC, 2016-05-10:
I wonder if the USB controller is locking up? Could you try creating a new profile in ckb with no animations to see if it still happens?

phorgan1, 2016-06-09:
Hello. I'm just checking in. With an uptodate as of today git pull, the problem is unchanged. The animations or no animations still locks up the keyboard.

phorgan1, 2016-06-17:
Why did you close this? I just reported that it still doesn't work with the latest git pull.

ccMSC, 2016-06-30:
Re-opening for now, but did you ever try creating the new profile as I asked?