ccMSC/ckb #317: k70 rgb doesn't work(osx 10.11.4, ckb 0.2.5)

ssissypsy, 2016-04-28:
First,thank you for your great work.Some issues puzzled me: when I haven't installed ckb on my mac, I could type by my k70, however, after I installed ckb(0.2.5 or 0.2.4), I couldn't type any letter with my k70, nothing happened no matter which key i pressing. Then I uninstall ckb, k70 works again, when I install ckb, k70 doesn't work.I have no idea about it.
Waiting for your response.

tatokis, 2016-04-28:
Assuming your K70 RGB is running firmware 2.04, there is a bug with it.
If you know how to compile ckb, it should be fixed in the fw2kb branch, otherwise please wait for a binary.
For more updates have a look at https://github.com/ccMSC/ckb/issues/298

ssissypsy, 2016-04-29:
you are riight, my k70 is running firmware 2.04.
thanks a lot

closed 2016-06-13 by ccMSC