ccMSC/ckb #318: addtional FN + F-keys combinations [on Strafe RGB]

john-2000, 2016-04-28:
Hi

firstly apologies if this is potentially very daft question, secondly thanks for all your work on this - without this I would have not bought a corsair.

finally my question:
my strafe keyboard has a function key and by default f5, f7 - f12 are used as media keys. It is of course possible to select multiple keys even in the binding screen but is assigning other key plus FN cominations to macros, program launchers or additional short cut keys like "xf86explorer" etc in spec for the current or future versions?

thanks

John

ccMSC, 2016-05-10:
It's definitely a planned feature. No ETA yet as I've been spending a lot of time on bug fixes.

john-2000, 2016-05-10:
GREAT, thanks so much.
Let me know if you there is something I can do to help.