ccMSC/ckb #319: K70 RGB with different Product ID

psychicbattle, 2016-04-29:
I'd had issues with the daemon not seeing any attached peripherals, and digging around in the code, I realized my keyboard actually has a different Product ID than you're looking for:

Relevant system_profiler snippet:

  Corsair Gaming K70 RGB RAPIDFIRE Keyboard :

        Product ID: 0x1b38
        Vendor ID: 0x1b1c  (CORSAIR MEMORY INC.)
        Version: 2.04
        Serial Number: 05020011AEAA1003550D7C87F5001940
        Speed: Up to 12 Mb/sec
        Manufacturer: Corsair
        Location ID: 0xfa120000 / 3
        Current Available (mA): 1000
        Current Required (mA): 500
        Extra Operating Current (mA): 0

In the existing codebase, K70RGBs are always 1b13.
I'm not sure if this is a newer model (did buy direct from manufacturer at a big event just this weekend) or if something else is going on.

When I update the definitions in usb.h and rebuild, the system recognizes the device, but errors out shortly thereafter:

sh-3.2# /Applications/ckb.app/Contents/Resources/ckb-daemon
    ckb: Corsair RGB driver beta-v0.2.5+f04
[I] Root controller ready at /var/run/ckb0
[I] Connecting Corsair Gaming K70 RGB RAPIDFIRE  at /var/run/ckb1
[I] Adding ckb1.ifusb[2]
[I] Connecting  at /var/run/ckb2
[I] Adding ckb2.ifhid[1]
[E] iterate_devices_hid (usb_mac.c:731): Failed to create device plugin: e00002be
[I] Adding ckb2.ifhid[0]
[E] iterate_devices_hid (usb_mac.c:731): Failed to create device plugin: e00002be
[E] iterate_devices_hid (usb_mac.c:731): Failed to create device plugin: e00002be
[E] iterate_devices_hid (usb_mac.c:731): Failed to create device plugin: e00002be

The keyboard becomes unresponsive until detached/reattached (the daemon doesn't attempt to reconnect to it unless the daemon is restarted at this point).

This is building off the fw2kb branch, on a 10.11.4 Mac, running the newest keyboard firmware (confirmed in the official windows software utility).

This looks like a different error than is being reported in #298, and the product_id issue definitely is, so I opted for making a new ticket.  I'll continue poking at the USB error in the meanwhile.

tatokis, 2016-04-29:
It appears that you have a k70 rgb rapidfire. 
This is a different model than the k70 rgb, and is probably not supported yet. 
 I assume some USB packet captures will help give more info on why it fails. Unfortunately I am not sure how that can be done on Mac.

Kithop, 2016-05-31:
Hi - I've actually just got one of these same keyboards; I've got access to Windows 7, OS X 10.11.5, and Xubuntu 16.04 (latter is where this keyboard is going to 'live' most of the time).  Anything I can do from here to help with data captures?

Kithop, 2016-05-31:
I grabbed 'USBPcap' for Windows (http://desowin.org/usbpcap/), shut down CUE, started a capture, and then fired up CUE again, flipping through the various built-in animations, hitting a few keys, and stopping the capture.

After looking at it quickly in Wireshark, I realise I should have done it on the 8ms/250Hz interrupt mode rather than 1ms/1000Hz (so... many... interrupts...).  You'll get my mouse (Razer Ouroboros) and whatever else mixed in I'm assuming, but here's the resulting packet capture!

[K70RGBRapidfire-pcap.zip](https://github.com/ccMSC/ckb/files/292393/K70RGBRapidfire-pcap.zip)

ccMSC, 2016-06-18:
Hi, there have been a lot of threads about new devices and I don't want to divide attention between all of them. I've opened up a new thread for them all: #386 - Please continue discussion there, thanks!

closed 2016-06-18 by ccMSC