ccMSC/ckb #320: K65 Lux RGB

joshholt, 2016-04-29:
This issue is related to #319 

I just purchased a K65 Lux RGB and I was finally able to get most things working through ckb except for the ability to apply lighting profiles.

The K65 Lux RGB properties are as follows:

Product ID:	0x1b37
Vendor ID:	0x1b1c  (CORSAIR MEMORY INC.)
Version:	2.04
Serial Number:	0500A034AEAA1003550D83D3F5001942
Speed:	Up to 12 Mb/sec
Manufacturer:	Corsair
Location ID:	0x14200000 / 18

I changed the ProductId in `usb.h` to match this instead of the original 0x1b17, I also update the fwupgradedialog file as stated in the comments. Once I did this ckb-daemon finally picked up my keyboard.

I have been able to successfully use ckb to redefine keys, but the lighting and animations do not work and the keyboard stays light with white light.

Here's the output from the daemon:

dev/MMXVI/ckb  fw2kb ✗                                                                                                                                                   
▶ sudo /Applications/ckb.app/Contents/Resources/ckb-daemon
    ckb: Corsair RGB driver beta-v0.2.5+f04
[I] Root controller ready at /var/run/ckb0
[I] Connecting Corsair Gaming K65 LUX RGB Keyboa at /var/run/ckb1
[I] Adding ckb1.ifusb[2]
[I] Adding ckb1.ifhid[1]
[I] Adding ckb1.ifhid[0]
[I] Starting input thread for /var/run/ckb1
[I] Reading ckb1.ifhid[0]
[I] Reading ckb1.ifhid[1]
[E] loadrgb_kb (led_keyboard.c:211): Bad input header
[W] _start_dev (device.c:45): Unable to load hardware profile
[I] Setup finished for /var/run/ckb1

owenzhao, 2016-05-09:
I have the same issue after upgrade from 1.13 firmware to 2.04 firmware.

esparkman, 2016-06-14:
@joshholt Could you send me your build that works with the K65 Lux?

ccMSC, 2016-06-18:
Hi, there have been a lot of threads about new devices and I don't want to divide attention between all of them. I've opened up a new thread for them all: #386 - Please continue discussion there, thanks!

closed 2016-06-18 by ccMSC

joshholt, 2016-08-04:
@esparkman sorry for the late reply on this, but I took the keyboard back to the store after another day fiddling with it after posting the issue and ended up with a p0k3r keyboard.