ccMSC/ckb #322: CKB won't recognize my k95 RGB

Dumbledorf, 2016-04-30:
I'm running 4.5.1-1-ARCH (Antergos) using GNOME Shell 3.20.1.

No matter what I do, CKB won't recognize my keyboard. I've tried different USB ports and re-installing CKB. None of it worked.

Jeffrey-Lin, 2016-05-01:
I'm having this issue on my K70 RGB as well. I'm using Mac OS X (El Capitan).

tatokis, 2016-05-01:
@Jeffrey-Lin You are probably having a different issue, check https://github.com/ccMSC/ckb/issues/298 out

Dumbledorf, 2016-05-09:
Any update on this issue?

ccMSC, 2016-05-10:
The firmware has recently been updated for the K95 as well, try the [`fw2kb`](https://github.com/ccMSC/ckb/tree/fw2kb) branch and see if you have any luck with it.

Dumbledorf, 2016-05-10:
No luck unfortunately.

Dumbledorf, 2016-05-11:
A few reboots fixed it for some reason.

closed 2016-05-11 by Dumbledorf