ccMSC/ckb #325: Strafe RGB not responding when ckb is on.

falconvk, 2016-05-02:
I dual-boot ubuntu with win10, and while in windows today my strafe had a firmware update.
Now as soon as ckb starts, the keyboard just stops responding.

Handsomfreddy, 2016-05-03:
Same issue here. Updated automatically in windows 10, strafe no longer functions with CKB in mac.  CKB is awesome I feel like someone cut off my arms without it, being that I can't adjust modifier keys in OS X : )

Thanks for any help.

Aimela, 2016-05-03:
I'm having the same issue with my K70 RGB on Linux Mint 17.3.
I wanted to be able to rebind the keys on my new Scimitar mouse(especially since bindings aren't saved to the device's memory like with the Logitech G600) and I'm not exactly up to firmware downgrading on my keyboard to achieve that.

Hopefully, the next update will come soon.

vanpelwc, 2016-05-04:
Same issue with K70 on OS X 10.11.4...  It's a good thing I looked before making a thread on this

tatokis, 2016-05-04:
People with K70 RGBs can have a look on https://github.com/ccMSC/ckb/issues/298 and compile the fw2kb branch which contains the fixes.
Not sure about the Strafe yet, but feel free to try the branch and see if it fixes the issue for you.

iainswarts, 2016-05-04:
Just tested the fw2kb branch on my Strafe RGB and it appears to be working perfectly! Thanks for the effort you guys put in :)

vanpelwc, 2016-05-04:
It works great! Thank you!

Handsomfreddy, 2016-05-04:
I compiled the fw2kb branch as well and it works great.  Not the easiest process to compile it though, not for the faint of heart.  But its working!!! 👍

Handsomfreddy, 2016-05-04:
@vanpelwc thanks for that tip, I was looking all over for where to change the modifiers!

Thanks!

closed 2016-06-13 by ccMSC