ccMSC/ckb #326: K70 RGB RAPIDFIRE?

rexlam1101, 2016-05-05:
Does it work with K70 RGB RAPIDFIRE? "No devices connected" :( It came with 2.04 as well, should be the same issue as #298 ? Does K70 RGB RAPIDFIRE use same firmware as K70 RGB?

tatokis, 2016-05-05:
Unfortunately it is not supported yet. 
If I recall correctly there is another report about this.

rexlam1101, 2016-05-05:
So basically K70 RGB RAPIDFIRE works differently with K70 RGB? :( I just bought it this weekend because I knew ckb exists :(

tatokis, 2016-05-05:
As mentioned in #319, it probably functions in a similar way, but different enough for ckb to not work with it. 
I don't have the hardware, so I can't really comment on it.
Will have to wait for @ccMSC 's input

Espionage724, 2016-05-10:
Reporting a me-too :p

I have the same vendor/product IDs reported in https://github.com/ccMSC/ckb/issues/319 (1b1c:1b38)

fourq, 2016-05-18:
Shit, so i'm in this same situation. What can I do to help get the ball rolling on getting the rapidfire working?

fourq, 2016-05-18:
so if you VM into a windows machine, install the corsair software and set your colors they will carry over b ack to linux/mac once you close the VM. However, the effects like ripple, wave, etc will not.

ccMSC, 2016-06-18:
Hi, there have been a lot of threads about new devices and I don't want to divide attention between all of them. I've opened up a new thread for them all: #386 - Please continue discussion there, thanks!

closed 2016-06-18 by ccMSC