ccMSC/ckb #327: Key Repeat Not Functioning - Corsair Strafe RGB

klaus385, 2016-05-05:
Try changing the default writes in OS X and no go i can get i+1 repeats but no other it will not go from i to n. When n is the number of characters.

klaus385, 2016-05-05:
This terminal command defaults write NSGlobalDomain KeyRepeat -int 0

mhennessie, 2016-05-06:
Key repeat doesn't work on my K65 RGB either.

klaus385, 2016-05-06:
Yeah hopefully it gets fixed its a really productivitilty killer. Don't get me wrong I'm thankful for this work! Just my only gripe.

ccMSC, 2016-05-10:
OS X has key repeat off by default and instead uses it as a means to insert special characters. Have you disabled this setting?

mhennessie, 2016-05-10:
@ccMSC that did the trick, thanks. For anyone else finding this here is the terminal command:

`defaults write -g ApplePressAndHoldEnabled -bool false`

closed 2016-06-13 by ccMSC