ccMSC/ckb #328: Can't bind Fn key on K65 RGB

mhennessie, 2016-05-05:
I would like to bind the Fn key as the right Option key and CKB lets me do that but it doesn't have any effect. The Fn key still functions as the Fn key. Additionally I would like to bind the Menu key as the Fn key. I can bind the Menu key as any other key but the Fn key. Is this possibly a hardware limitation? Right now I have the Menu key bound as the right Option key.

My goal is to swap the Fn and Option keys in this pic so it is like a real Apple keyboard layout:

[image]

FYI: I am running CKB 0.2.5+f04 fw2kb with firmware version 2.04.

ccMSC, 2016-05-10:
I don't think it's possible to move the Fn functionality elsewhere as it seems to be enforced by the keyboard hardware. I don't own this exact model so somebody who does might have more information.

mhennessie, 2016-05-10:
That's what I was afraid of and I figured I might have to live with it. If there is any type of dump I can provide to give any insight let me know and I will do my best to provide that.

closed 2016-06-13 by ccMSC