ccMSC/ckb #330: Strafe RGB not recognized as connecting

audreyseo, 2016-05-06:
This is probably just a minor bug. Every so often, if I unplug the keyboard from my laptop, which I connect to a larger display on my desktop, then CKB won't recognize it and I will have to re-install CKB all over again. I'm not really sure how this happens or why. It happened once when I put the laptop to sleep and unplugged everything with the laptop closed. Recently, it happened even after I opened my laptop up and waited a few seconds last night. When I went to plug in my strafe, the keys only turned white, and I'm pretty sure that it was not responding. Once I reinstall CKB, it works, and the keys are animated. I'm using a mac, yosemite.

mhennessie, 2016-05-06:
When it isn't detected try launching Activity Monitor and force quit the CKB Daemon. The daemon should automatically restart and it might detect your keyboard then.

jameskrawczyk, 2016-05-08:
Try reverting to a previous version of ckb. That's what worked for me and the Strafe RGB.

audreyseo, 2016-05-19:
@mhennessie, thank you, this worked for me!

closed 2016-05-19 by audreyseo