ccMSC/ckb #331: ckb is non responsive the latest k95 firmware update

z1lt0id, 2016-05-07:
When on I was on my Windows partition today it updated the firmware on my K95 RGB.  Later on the in the day go back into my Linux partition to find that the keyboard is non responsive and that CKB is unable to load any hardware profile.  Below is the output from ckb-daemon and I'm running 0.2.5beta.
````
[I] Root controller ready at /dev/input/ckb0
[I] Connecting Corsair K95 RGB Gaming Keyboard at /dev/input/ckb1
[I] Starting input thread for /dev/input/ckb1
[E] loadrgb_kb (led_keyboard.c:211): Bad input header
[W] _start_dev (device.c:45): Unable to load hardware profile
[I] Setup finished for /dev/input/ckb1
[E] os_usbsend (via device_keyboard.c:38): No such file or directory
[I] Attempting reset...
[E] os_usbsend (via device_keyboard.c:38): No such file or directory
[E] os_usbsend (via device_keyboard.c:38): No such file or directory
[E] os_usbsend (via device_keyboard.c:38): No such file or directory
[E] os_resetusb (via usb.c:169): resetusb failed: No such device
[E] usb_tryreset (usb.c:177): Reset failed. Disconnecting.
[I] Disconnecting /dev/input/ckb1
[I] Removed device path /dev/input/ckb1

[I] Caught signal 15
[I] Closing root controller
[I] Removed device path /dev/input/ckb0
````

fliiiix, 2016-05-07:
Have your tired the fw2kb branch?

z1lt0id, 2016-05-07:
No, I'll check it out now. Thanks for the suggestion

z1lt0id, 2016-05-07:
Unfortunately that branch didn't either.  Here is the output of ckb-daemon
````
    ckb: Corsair RGB driver beta-v0.2.5
[I] Root controller ready at /dev/input/ckb0
[I] Connecting Corsair K95 RGB Gaming Keyboard at /dev/input/ckb1
[I] Starting input thread for /dev/input/ckb1
[E] loadrgb_kb (led_keyboard.c:211): Bad input header
[W] _start_dev (device.c:45): Unable to load hardware profile
[I] Setup finished for /dev/input/ckb1
[E] os_usbsend (via device_keyboard.c:38): No such file or directory
[I] Attempting reset...
[E] os_usbsend (via device_keyboard.c:38): No such file or directory
[E] os_usbsend (via device_keyboard.c:38): No such file or directory
[E] os_usbsend (via device_keyboard.c:38): No such file or directory
[E] os_usbsend (via device_keyboard.c:38): No such file or directory
[E] os_usbsend (via device_keyboard.c:38): No such file or directory
[E] os_usbsend (via device_keyboard.c:38): No such file or directory
[E] os_usbsend (via device_keyboard.c:38): No such file or directory
[E] os_usbsend (via device_keyboard.c:38): No such file or directory
[E] os_usbsend (via device_keyboard.c:38): No such file or directory
[E] os_usbsend (via device_keyboard.c:38): No such file or directory
[E] os_usbsend (via device_keyboard.c:38): No such file or directory
[E] os_usbsend (via device_keyboard.c:38): No such file or directory
[E] os_usbsend (via device_keyboard.c:38): No such file or directory
[E] os_usbsend (via device_keyboard.c:38): No such file or directory
[E] os_usbsend (via device_keyboard.c:38): No such file or directory
[E] os_usbsend (via device_keyboard.c:38): No such file or directory
[E] os_usbsend (via device_keyboard.c:38): No such file or directory
[E] os_usbsend (via device_keyboard.c:38): No such file or directory
[E] os_usbsend (via device_keyboard.c:38): No such file or directory
[E] os_usbsend (via device_keyboard.c:38): No such file or directory
[E] os_usbsend (via device_keyboard.c:38): No such file or directory
[E] os_usbsend (via device_keyboard.c:38): No such file or directory
[E] os_usbsend (via device_keyboard.c:38): No such file or directory
[E] os_usbsend (via device_keyboard.c:38): No such file or directory
[E] os_usbsend (via device_keyboard.c:38): No such file or directory
[E] os_usbsend (via device_keyboard.c:38): No such file or directory
[E] os_usbsend (via device_keyboard.c:38): No such file or directory
[E] os_usbsend (via device_keyboard.c:38): No such file or directory
[E] os_usbsend (via device_keyboard.c:38): No such file or directory
[E] os_usbsend (via device_keyboard.c:38): No such file or directory
[E] os_usbsend (via device_keyboard.c:38): No such file or directory
[E] os_usbsend (via device_keyboard.c:38): No such file or directory
[E] os_resetusb (via usb.c:169): resetusb failed: No such device
[E] usb_tryreset (usb.c:177): Reset failed. Disconnecting.
[I] Disconnecting /dev/input/ckb1
[I] Removed device path /dev/input/ckb1
[I] Connecting  at /dev/input/ckb1
[W] os_setupusb (usb_linux.c:292): Unable to read endpoint count from udev, assuming 4...
[E] os_setupusb (usb_linux.c:295): Failed to claim interfaces: No route to host
[I] Disconnecting /dev/input/ckb1
[I] Removed device path /dev/input/ckb1
````

tatokis, 2016-05-07:
I recall ccMSC saying that the firmware upgrade came out for the K70 RGB first, and the K95 RGB didn't get one (until recently, as it looks like).
So there are probably some minor differences. Will have to wait for it to get fixed.

Until then, you can downgrade your firmware on windows in order to restore functionality.
Here's a thread with past firmwares http://forum.corsair.com/v3/showthread.php?t=138472

z1lt0id, 2016-05-07:
I figured that would be the case, shouldn't have updated it :( Thanks
for your help either way.

ccMSC, 2016-05-10:
@z1lt0id the driver log indicates that it's still running the `master` branch, try redownloading and recompiling?

I was able to download the new K95 firmware today and confirmed that the `fw2kb` branch is in fact working with it, at least on my machine.

z1lt0id, 2016-05-10:
Ahh cheers, not sure why it was still running on the master branch. I'll try again on the fw2kb branch.

z1lt0id, 2016-05-10:
@ccMSC just to let you know I succesfully compiled the fw2kb branch and working correctly.

closed 2016-05-10 by z1lt0id