ccMSC/ckb #333: Raspberry PI deadlocks during compile

denali1, 2016-05-08:
It doesn't have enough memory or power to spin up 24 compile jobs.  quickinstall has to be modified to change the make -j 24 to make -j 1.  Ultimately, it would be nice to see a command line param for quickinstall so the number of jobs can be set going in, rather than having to edit the script.

tatokis, 2016-05-08:
I think replacing -j 24 with -j `getconf _NPROCESSORS_ONLN` would be more appropriate in this case, as it prints the number of online cores so that it starts up the same number of threads.

denali1, 2016-05-08:
My only concern with that lies within the Pi itself.  The 2B has 1G of RAM on a dual core 1gHz ARM processor.  It uses shared memory for video, so it actually has less than 1G of RAM before the OS loads.  The Pi 3 might be able to get away with that, but I have my doubts about the older models.  I'll make the mod and try it just for science's sake, then report back.

denali1, 2016-05-08:
I'm the kind of person to admit when I was wrong.  That did, indeed, work.  It's a good solution to the issue.

closed 2016-06-13 by ccMSC