ccMSC/ckb #334: CKB under Raspberry PI Issues

denali1, 2016-05-08:
1) ckb rebinds mousewheel down as mousewheel up.  No amount of reseting the bind fixes it.
2) Animations do not run.  I have a rainbow wave setup to run indefinitely, but it doesn't actually execute.

tatokis, 2016-05-08:
Regarding 2, the RPi seems to have a max limit of 1A for the USB, which should be enough for the keyboard, so make sure your power supply can provide more than 1A.

Other than that, please make sure that your device functions on a normal desktop with CKB, in order to make sure that this is a Pi specific bug. (As I don't see how it could be.)

It could just be that corsair released a new firmware for it and broke things again, just like they did with most of the keyboards. If you upgraded its firmware recently, you could try downgrading it and see if that fixes it.

denali1, 2016-05-08:
I'll make the mod to the Pi that ups USB port amperage to 1A and see what happens.

It works fine under Windows 10 (AMD 1100T, 16GB RAM).  I actually had to move it to my laptop (Windows 7, AMD Mobile chip, 8GB RAM) to test it after my Windows 10 unit flamed out.  I retested it this morning and verified proper operation.  Mousewheel went back to it's proper operation (up is up and down is down) and lights acted as expected.

A point of irony, the programming that I saved to it with ckb fired up and started running under CUE.  CUE (1.64.12) and Firmware (2.03) are the latest and has not changed in the last three weeks or so (IIRC).

denali1, 2016-05-08:
Upping the USB port amperage to 1.2A from 600mA made no change.  The P/S provides 2.5A to the board.

denali1, 2016-05-08:
Okay, odd thing I've noticed.  The light that's usually mapped to indicate what DPS the mouse is set at?  ckb will change it on the RPi.  That's the only one it seems to be able to touch.

ccMSC, 2016-05-10:
Out of curiosity have you tried fiddling with the frame rate setting? Some cheaper USB ports seem to have issues with it and can get locked up even at 30fps. It would also be helpful if you could provide the output from the daemon to see if it's spitting any errors.

denali1, 2016-05-10:
I haven't.  I'll see about trying that.  Pertinent portion of Syslog shown below.

`denali@Ezerhardin:~$ cat /var/log/syslog | grep "ckb"
May  9 20:21:48 Ezerhardin ckb-daemon[367]:     ckb: Corsair RGB driver beta-v0.2.5
May  9 20:21:48 Ezerhardin ckb-daemon[367]: [I] Root controller ready at /dev/input/ckb0
May  9 20:21:51 Ezerhardin ckb-daemon[367]: [I] Connecting Corsair Gaming Scimitar RGB Mouse at /dev/input/ckb1
May  9 20:21:51 Ezerhardin ckb-daemon[367]: [I] Starting input thread for /dev/input/ckb1
May  9 20:21:51 Ezerhardin kernel: [   12.695608] input: ckb1: Corsair Gaming Scimitar RGB Mouse as /devices/virtual/input/input2
May  9 20:21:51 Ezerhardin kernel: [   12.697204] input: ckb1: Corsair Gaming Scimitar RGB Mouse as /devices/virtual/input/input3
May  9 20:21:56 Ezerhardin ckb-daemon[367]: [E] os_usbrecv (via firmware.c:15): Connection timed out
May  9 20:21:56 Ezerhardin ckb-daemon[367]: [W] _start_dev (device.c:24): Unable to load firmware version/poll rate
May  9 20:21:56 Ezerhardin ckb-daemon[367]: [I] Setup finished for /dev/input/ckb1
May  9 20:21:56 Ezerhardin kernel: [   17.803276] usb 1-1.3: usbfs: USBDEVFS_CONTROL failed cmd ckb-daemon rqt 161 rq 1 len 64 ret -110`