ccMSC/ckb #335: Strafe RGB unresponsive after few key presses

Dmitrev, 2016-05-11:
While CKB is running, right after i insert my keyboard, i can press maybe 4-5 characters before it stops responding. CKB will strill recognize that my keyboard is connected, but it cannot control the LEDs

Here is the output from `dmesg | tail -400`

```
dmesg | tail -400
[  861.696321] xhci_hcd 0000:00:14.0: xHCI xhci_drop_endpoint called with disabled ep ffff88009d67bd80
[  861.696326] xhci_hcd 0000:00:14.0: xHCI xhci_drop_endpoint called with disabled ep ffff88009d67b4e0
[  861.696328] xhci_hcd 0000:00:14.0: xHCI xhci_drop_endpoint called with disabled ep ffff88009d67b6c0
[  862.072305] usb 1-3: reset full-speed USB device number 16 using xhci_hcd
[  862.200874] xhci_hcd 0000:00:14.0: xHCI xhci_drop_endpoint called with disabled ep ffff88009d67bd80
[  862.200880] xhci_hcd 0000:00:14.0: xHCI xhci_drop_endpoint called with disabled ep ffff88009d67b4e0
[  862.200882] xhci_hcd 0000:00:14.0: xHCI xhci_drop_endpoint called with disabled ep ffff88009d67b6c0
[  862.576814] usb 1-3: reset full-speed USB device number 16 using xhci_hcd
[  862.705454] xhci_hcd 0000:00:14.0: xHCI xhci_drop_endpoint called with disabled ep ffff88009d67bd80
[  862.705473] xhci_hcd 0000:00:14.0: xHCI xhci_drop_endpoint called with disabled ep ffff88009d67b4e0
[  862.705475] xhci_hcd 0000:00:14.0: xHCI xhci_drop_endpoint called with disabled ep ffff88009d67b6c0
[  862.705823] usb 1-3: usbfs: process 13466 (ckb-daemon) did not claim interface 0 before use
[  863.081459] usb 1-3: reset full-speed USB device number 16 using xhci_hcd
[  863.210009] xhci_hcd 0000:00:14.0: xHCI xhci_drop_endpoint called with disabled ep ffff88009d67bd80
[  863.210018] xhci_hcd 0000:00:14.0: xHCI xhci_drop_endpoint called with disabled ep ffff88009d67b4e0
[  863.210022] xhci_hcd 0000:00:14.0: xHCI xhci_drop_endpoint called with disabled ep ffff88009d67b6c0
[  863.586061] usb 1-3: reset full-speed USB device number 16 using xhci_hcd
[  863.714647] xhci_hcd 0000:00:14.0: xHCI xhci_drop_endpoint called with disabled ep ffff88009d67bd80
[  863.714652] xhci_hcd 0000:00:14.0: xHCI xhci_drop_endpoint called with disabled ep ffff88009d67b4e0
[  863.714655] xhci_hcd 0000:00:14.0: xHCI xhci_drop_endpoint called with disabled ep ffff88009d67b6c0
[  864.090652] usb 1-3: reset full-speed USB device number 16 using xhci_hcd
[  864.219249] xhci_hcd 0000:00:14.0: xHCI xhci_drop_endpoint called with disabled ep ffff88009d67bd80
[  864.219267] xhci_hcd 0000:00:14.0: xHCI xhci_drop_endpoint called with disabled ep ffff88009d67b4e0
[  864.219269] xhci_hcd 0000:00:14.0: xHCI xhci_drop_endpoint called with disabled ep ffff88009d67b6c0
[  864.595129] usb 1-3: reset full-speed USB device number 16 using xhci_hcd
[  864.723824] xhci_hcd 0000:00:14.0: xHCI xhci_drop_endpoint called with disabled ep ffff88009d67bd80
[  864.723831] xhci_hcd 0000:00:14.0: xHCI xhci_drop_endpoint called with disabled ep ffff88009d67b4e0
[  864.723835] xhci_hcd 0000:00:14.0: xHCI xhci_drop_endpoint called with disabled ep ffff88009d67b6c0
[  865.099763] usb 1-3: reset full-speed USB device number 16 using xhci_hcd
[  865.228410] xhci_hcd 0000:00:14.0: xHCI xhci_drop_endpoint called with disabled ep ffff88009d67bd80
[  865.228416] xhci_hcd 0000:00:14.0: xHCI xhci_drop_endpoint called with disabled ep ffff88009d67b4e0
[  865.228418] xhci_hcd 0000:00:14.0: xHCI xhci_drop_endpoint called with disabled ep ffff88009d67b6c0
[  865.604359] usb 1-3: reset full-speed USB device number 16 using xhci_hcd
[  865.732871] xhci_hcd 0000:00:14.0: xHCI xhci_drop_endpoint called with disabled ep ffff88009d67bd80
[  865.732877] xhci_hcd 0000:00:14.0: xHCI xhci_drop_endpoint called with disabled ep ffff88009d67b4e0
[  865.732881] xhci_hcd 0000:00:14.0: xHCI xhci_drop_endpoint called with disabled ep ffff88009d67b6c0
[  866.108949] usb 1-3: reset full-speed USB device number 16 using xhci_hcd
[  866.237588] xhci_hcd 0000:00:14.0: xHCI xhci_drop_endpoint called with disabled ep ffff88009d67bd80
[  866.237594] xhci_hcd 0000:00:14.0: xHCI xhci_drop_endpoint called with disabled ep ffff88009d67b4e0
[  866.237597] xhci_hcd 0000:00:14.0: xHCI xhci_drop_endpoint called with disabled ep ffff88009d67b6c0
[  866.613457] usb 1-3: reset full-speed USB device number 16 using xhci_hcd
[  866.742029] xhci_hcd 0000:00:14.0: xHCI xhci_drop_endpoint called with disabled ep ffff88009d67bd80
[  866.742035] xhci_hcd 0000:00:14.0: xHCI xhci_drop_endpoint called with disabled ep ffff88009d67b4e0
[  866.742037] xhci_hcd 0000:00:14.0: xHCI xhci_drop_endpoint called with disabled ep ffff88009d67b6c0
[  867.118051] usb 1-3: reset full-speed USB device number 16 using xhci_hcd
[  867.246607] xhci_hcd 0000:00:14.0: xHCI xhci_drop_endpoint called with disabled ep ffff88009d67bd80
[  867.246617] xhci_hcd 0000:00:14.0: xHCI xhci_drop_endpoint called with disabled ep ffff88009d67b4e0
[  867.246622] xhci_hcd 0000:00:14.0: xHCI xhci_drop_endpoint called with disabled ep ffff88009d67b6c0
[  867.622642] usb 1-3: reset full-speed USB device number 16 using xhci_hcd
[  867.751303] xhci_hcd 0000:00:14.0: xHCI xhci_drop_endpoint called with disabled ep ffff88009d67bd80
[  867.751310] xhci_hcd 0000:00:14.0: xHCI xhci_drop_endpoint called with disabled ep ffff88009d67b4e0
[  867.751314] xhci_hcd 0000:00:14.0: xHCI xhci_drop_endpoint called with disabled ep ffff88009d67b6c0
[  868.127158] usb 1-3: reset full-speed USB device number 16 using xhci_hcd
[  868.255865] xhci_hcd 0000:00:14.0: xHCI xhci_drop_endpoint called with disabled ep ffff88009d67bd80
[  868.255872] xhci_hcd 0000:00:14.0: xHCI xhci_drop_endpoint called with disabled ep ffff88009d67b4e0
[  868.255874] xhci_hcd 0000:00:14.0: xHCI xhci_drop_endpoint called with disabled ep ffff88009d67b6c0
[  868.466923] usb 1-3: usbfs: process 13466 (ckb-daemon) did not claim interface 0 before use
[  868.631800] usb 1-3: reset full-speed USB device number 16 using xhci_hcd
[  868.760292] xhci_hcd 0000:00:14.0: xHCI xhci_drop_endpoint called with disabled ep ffff88009d67bd80
[  868.760299] xhci_hcd 0000:00:14.0: xHCI xhci_drop_endpoint called with disabled ep ffff88009d67b4e0
[  868.760303] xhci_hcd 0000:00:14.0: xHCI xhci_drop_endpoint called with disabled ep ffff88009d67b6c0
[  868.779945] input: Corsair Corsair STRAFE RGB Gaming Keyboard as /devices/pci0000:00/0000:00:14.0/usb1/1-3/1-3:1.0/0003:1B1C:1B20.0034/input/input81
[  868.836020] hid-generic 0003:1B1C:1B20.0034: input,hidraw0: USB HID v1.11 Keyboard [Corsair Corsair STRAFE RGB Gaming Keyboard] on usb-0000:00:14.0-3/input0
[  869.276521] usb 1-3: reset full-speed USB device number 16 using xhci_hcd
[... 312 lines cut ...]
[ 1419.405482] usb 1-3: New USB device found, idVendor=1b1c, idProduct=1b20
[ 1419.405490] usb 1-3: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[ 1419.405494] usb 1-3: Product: Corsair STRAFE RGB Gaming Keyboard
[ 1419.405497] usb 1-3: Manufacturer: Corsair
[ 1419.405499] usb 1-3: SerialNumber: 0503D015AEAA1002550458E4F5001943
[ 1419.407287] input: Corsair Corsair STRAFE RGB Gaming Keyboard as /devices/pci0000:00/0000:00:14.0/usb1/1-3/1-3:1.0/0003:1B1C:1B20.003E/input/input91
[ 1419.459899] hid-generic 0003:1B1C:1B20.003E: input,hidraw0: USB HID v1.11 Keyboard [Corsair Corsair STRAFE RGB Gaming Keyboard] on usb-0000:00:14.0-3/input0
[ 1429.471055] hid-generic 0003:1B1C:1B20.003F: usb_submit_urb(ctrl) failed: -1
[ 1429.471079] hid-generic 0003:1B1C:1B20.003F: timeout initializing reports
[ 1429.471818] input: Corsair Corsair STRAFE RGB Gaming Keyboard as /devices/pci0000:00/0000:00:14.0/usb1/1-3/1-3:1.1/0003:1B1C:1B20.003F/input/input92
[ 1429.527560] hid-generic 0003:1B1C:1B20.003F: input,hiddev0,hidraw2: USB HID v1.11 Keyboard [Corsair Corsair STRAFE RGB Gaming Keyboard] on usb-0000:00:14.0-3/input1
[ 1429.527685] usbhid 1-3:1.2: couldn't find an input interrupt endpoint
[ 1460.918260] usb 1-1: USB disconnect, device number 12
[ 1503.187070] usb 1-4: new full-speed USB device number 19 using xhci_hcd
[ 1503.316611] usb 1-4: New USB device found, idVendor=1532, idProduct=0034
[ 1503.316618] usb 1-4: New USB device strings: Mfr=1, Product=2, SerialNumber=0
[ 1503.316622] usb 1-4: Product: Razer Taipan
[ 1503.316625] usb 1-4: Manufacturer: Razer
[ 1503.318189] input: Razer Razer Taipan as /devices/pci0000:00/0000:00:14.0/usb1/1-4/1-4:1.0/0003:1532:0034.0040/input/input93
[ 1503.318466] hid-generic 0003:1532:0034.0040: input,hidraw1: USB HID v1.11 Mouse [Razer Razer Taipan] on usb-0000:00:14.0-4/input0
[ 1503.320144] input: Razer Razer Taipan as /devices/pci0000:00/0000:00:14.0/usb1/1-4/1-4:1.1/0003:1532:0034.0041/input/input94
[ 1503.427782] hid-generic 0003:1532:0034.0041: input,hidraw3: USB HID v1.11 Keyboard [Razer Razer Taipan] on usb-0000:00:14.0-4/input1
[ 1503.428938] input: Razer Razer Taipan as /devices/pci0000:00/0000:00:14.0/usb1/1-4/1-4:1.2/0003:1532:0034.0042/input/input95
[ 1503.483797] hid-generic 0003:1532:0034.0042: input,hidraw4: USB HID v1.11 Keyboard [Razer Razer Taipan] on usb-0000:00:14.0-4/input2
```

Dmitrev, 2016-05-11:
I think this issue is related to the recent update of the CUE software on Windows. It might have updated the firmware inside the Keyboard.

tatokis, 2016-05-11:
If that is the case, try compiling the fw2kb branch and see if that 
fixes anything.

mhennessie, 2016-05-12:
Other have reported that reverting to the v0.2.4 build of CKB fixes this issue.

Dmitrev, 2016-05-12:
I will try both solutions after work

bgidley, 2016-05-13:
I have the same issue - I've tried 0.2.4 and it hasn't helped. Will now try the branch.

bgidley, 2016-05-13:
The branch works fine with the updated firmware.

Tazmain, 2016-05-14:
The sabre branch works for me

closed 2016-06-13 by ccMSC