ccMSC/ckb #337: Strafe firmware upgrade to 1.33 failed

cosmozhang, 2016-05-11:
When I was trying to upgrade the firmware to 1.33 from 1.30, an error msg shows as "There was a problem with the downloaded file, please try again later". Can we manually download the new firmware and place it at some designated place instead? Thank you!

tatokis, 2016-05-11:
Since corsair released some new firmwares recently that break support with ckb, I would not recommend updating the firmware yet, until we can be sure all devices are functioning properly. 
Ckb downloads the firmware from corsair, and the latest version is probably greater than 1.33, but ckb is not aware of that yet, which could explain why it fails.

cosmozhang, 2016-05-11:
You are right. Just I upgraded it to 1.33 from Windows and reconnect it to ubuntu, it is not working at all..(Press any key no response). Shall I downgrade it to 1.30? Or will there be any fix soon? 
Thank you!

tatokis, 2016-05-11:
There is a fix for some keyboards in the fw2kb branch. More details are in #298 about that. 
If it doesn't work for you, you can downgrade. 
I don't have a strafe myself, and I am not sure what keyboards got the updates, so apologies if this doesn't apply to the strafe.

cosmozhang, 2016-05-11:
Thank you. This branch works for me now!

closed 2016-06-13 by ccMSC