ccMSC/ckb #338: F1-F12 compatibility with Karabiner

owenzhao, 2016-05-12:
Karabiner is a keyboard remapping tool for OS X. When I am using another keyboard, Cherry MX 6.0, I am using Karabiner to remap F1-F12 to common mac keys. (F1-F2 to turn down and up screen lightness, F3:Launchpad, F4:Mission Control, F7:Previous, F8:Play/Pause, F9:Next, F10:Mute, F11:Volume Down, F12:Volume Up).

It is useful as the keyboard default setting Fn+F9 to Fn+F12 only Fn+F11 works, the others don't work. (Mine is K65 RGB). 

I find that, if I connect my keyboard with Windows (Windows 10 in Parallels Desktop), set the K65 to Windows 10, using the official utilities. Then disconnect K65 from Windows 10. Karabiner will work. However, ckb will not find K65 any more and the light pattern doesn't work.

However, if ckb works, Carabiner will not work. It can detect F1 to F12 events, but it can not remap them.

I want to know if I could use ckb with Karabiner together. As multimedia keys like stop, previous and next, ckb's remapping do not work. Karabiner's work.

ccMSC, 2016-06-13:
Karabiner has come up before and AFAIK it's not possible to make compatible with ckb. Has to do with the way Apple handles keyboard drivers. :/

owenzhao, 2016-06-13:
If that is not possible, how about emulate the keystroke as it does?

ccMSC, 2016-06-30:
It's a planned feature. Don't know yet when it will be done.

kashura, 2017-05-17:
@ccMSC Hi, so currently there is no way to map F keys to mission control or launchpad?