ccMSC/ckb #340: ckb causing airportd to perform wifi scan - causes ping spikes

vertiman, 2016-05-14:
This is a weird bug.  It looks like ckb causes airportd to do wifi scans, and wifi scans cause temporary ping spikes.  This happens every 5-10 seconds in Console.app.

2016-05-14 12:37:42.915 AM airportd[1759]: com.apple.message.domain: com.apple.wifi.internal.scanning
com.apple.message.error: No Error
com.apple.message.duration: 0
com.apple.message.process_name: ckb
com.apple.message.mac_address: a0:99:9b:xx:xx:xx
SenderMachUUID: D1FCF015-F884-34A4-87C2-5D94E1EBC21D

I can confirm that killing ckb makes the airportd process stop scanning every 5 seconds, and that causes my ping spikes to disappear.  Very weird.

Any other information that would be helpful, let me know!

ccMSC, 2016-06-13:
I'm guessing it's the firmware update check. If you turn it off (from the Settings screen) then the spikes should go away.

bboysnick, 2016-12-05:
Having the same issue here. Turning off the automatic update won't help.

vertiman, 2016-12-05:
Sorry to say that only way I solved this was removing ckb.  It's not needed in Sierra unless you want to use the lights, and the project seems to be abandoned at this point.