ccMSC/ckb #341: OS X 10.11.4 K65 Not working

anthonycastelli, 2016-05-14:
I just installed all of this and even tried compiling it myself but have had no luck getting my keyboard to work. I've updated to the latest version (2.04) and its still not working. Is that firmware too new for this? I'm tried a bunch of stuff and can't seem to get it to work. Any help is greatly appreciated!

tatokis, 2016-05-14:
If you are running firmware 2.04, make sure to compile the fw2kb branch instead.

anthonycastelli, 2016-05-14:
I have tried that and still no luck.

tatokis, 2016-05-14:
Do you happen to have a K65 Lux RGB? (#320)

anthonycastelli, 2016-05-14:
No, its the K65 RGB, the regular one I guess.

anthonycastelli, 2016-05-14:
Still no luck with the keyboard. I don't even get it to recognize it. Ive tried a few things but no dice. Any other ideas?

ccMSC, 2016-06-18:
Hi, there have been a lot of threads about new devices and I don't want to divide attention between all of them. I've opened up a new thread for them all: #386 - Please continue discussion there, thanks!

closed 2016-06-18 by ccMSC