ccMSC/ckb #342: level meters - triggered animations ?

john-2000, 2016-05-14:
Hi,

more of a feature request / question -- and your main plan forward is much more important than this.

I was just wondering if it would be possible to trigger animations via some kind of programmable interface. The thing that popped into my head first would be a using the for f-keys as level meters. 

http://img04.deviantart.net/715d/i/2013/329/6/e/vu_level_meter_by_lewelsch-d6vl192.png

Anyway this could be done? Or anyway we/I could help with that?

thx

ccMSC, 2016-06-18:
I never got around to documenting the animation interface, but they are just normal programs (you could also write one in python) so that might be able to do what you want. I'd like to introduce a command-line interface to allow more programmability within the application, but it's probably not going to be available for a while.

john-2000, 2016-06-18:
> 
I never got around to documenting the animation interface, but they are just normal programs (you could also write one in python) ...

excuse my real beginner's status but I believe the current programmes are written in c(?) how could I use python to interact properly with ckb ?

(( PS I know you have a lot of things you are working on for this so I appreciate that all the effort you put in ))

ccMSC, 2016-06-30:
They are, you'd basically have to read over the C code and convert it into python. I'm hoping to implement proper python bindings at some point but as I said, I couldn't give a timeline.

grounded042, 2016-06-30:
If I remember correctly from a while back, you should just be able to write to `/tmp/ckb1/cmd` using python. A while back I started working on something to do that and just haven't had time to work on it: https://github.com/grounded042/pyckb