ccMSC/ckb #344: Corsair K65 RGB Rapidfire not being recognized

rbieb, 2016-05-15:
Ckb says
    No devices connected

I already saw that the rapidfire isn't being supported yet, but I suppose opening an issue could channel all the info towards getting it to work in one place.

newnewcorsair, 2016-05-16:
same here. Using on Mac 10.11.4

edman545, 2016-05-25:
Not working on xubuntu 15.10 here.

Kithop, 2016-06-12:
Yeah, I pulled a USB packet capture on my K70 Rapidfire in #319 , from under Windows with the CUE software running.  Suggest you guys follow that one, or if anyone wants to try their hand at reading it as I'm not a dev... :)

ccMSC, 2016-06-18:
Hi, there have been a lot of threads about new devices and I don't want to divide attention between all of them. I've opened up a new thread for them all: #386 - Please continue discussion there, thanks!

closed 2016-06-18 by ccMSC