ccMSC/ckb #348: OS X 10.11.5 can not detect the newest K70 Rapidfire.

yuxuyang, 2016-05-17:
I tried all the way to connect this keyboard.. But still not working. and i am not sure how to find the log info. please help

yuxuyang, 2016-05-18:
So I changed some code in the v0.2.5. and it works, the keyboard can be detected. But the problem is after detected, the input doesn't work. Anyone has any idea to help?

tatokis, 2016-05-18:
#319 Rapidfire is not supported, probably requires changes in the input code as well.

yuxuyang, 2016-05-18:
I figure out a way to deal with the input thing by restart my Mac and find that the keyboard now is useful but you can not do the lighting stuff. The lighting menu bar disappear. Anyone can point me the the right spot for the lighting code?

fourq, 2016-05-18:
so if you VM into a windows machine, install the corsair software and set your colors they will carry over b ack to linux/mac once you close the VM. However, the effects like ripple, wave, etc will not.

yuxuyang, 2016-05-19:
I tried that, actually, that didn't help with anything... And the question is when would we have the update staff..

CorreAktor, 2016-06-14:
+1 on support. Using VMware is a great temporary fix to change the settings but would be awesome if ckb is updated

ccMSC, 2016-06-18:
Hi, there have been a lot of threads about new devices and I don't want to divide attention between all of them. I've opened up a new thread for them all: #386 - Please continue discussion there, thanks!

closed 2016-06-18 by ccMSC

elbuenyuz, 2016-09-29:
i have the same problem, i just bought it i have a mac , if someone make it work please share the knowledge :) .... im gonna try the VM for now!