ccMSC/ckb #349: DPI switching suddenly stopped working on on single Scimitar mode.

Aimela, 2016-05-18:
So, yesterday DPI, switching suddenly stopped working on a single mode for my scimitar mouse. All I was doing was switching the DPI up and down and then it became stuck on the lowest DPI setting. I tried restarting the program and my PC to no effect. Changing the DPI button bindings had no effect, either.

Thing is, only that one mode is affected. If I duplicate the mode, it is affected as well. Nothing else is affected.

I am using the fw2kb branch on Linux Mint 17.3 with both a Scimitar and a K70 RGB.

ccMSC, 2016-06-13:
Could you upload your `~/.config/ckb/ckb.conf` file? Should have the information to tell me why it's stuck.

Aimela, 2016-06-13:
Sorry, I've installed a new distro over since this happened and didn't take that config with me. However, if I can get this to happen again, I'll upload the config file.

gtjoseph, 2016-06-18:
Same issue.  Just got the mouse today and it was super speedy so I set buttons 10 and 12 to DPI Down and DPI Up.   I hit down a few times and the DPI dropped to 800x800 and nothing I've done since has changed it.  I've tried setting the buttons to specific stages and even a custom  setting.  The only behavior change I've seen is that when I press a button assigned a custom DPI, the DPI indicator changes while I'm pressing the button then goes back to the default color.  Tried both master and newdev branches.

[ckb.conf.txt](https://github.com/ccMSC/ckb/files/322036/ckb.conf.txt)