ccMSC/ckb #350: Tick/Tilde not binding properly

brettnem, 2016-05-19:
Hello,
I'm using a Corsair K70 with El Cap. I have other key bindings working just fine. The "tick/tilde" key (which I cannot easily type because of this problem) just isn't working. I clearly see
(tick symbol) --> (tick symbol) in the bindings:
https://www.evernote.com/l/AAPipRI23JJLEproSRuY6cyNrcV49-sTc90B/image.png

But when I push that key, I get this guy: § <-- actually pushed it (ascii code 245).

If I bind it to something else, like say, the letter "a" it works and prints an "a". So I just can't seem to get the tick key working.

And in case it helps shift+(tick key) gives me this: ± (ascii code 241)

Running version ckb beta v0.2.2. Actually, I just upgraded to ckb beta v0.2.5 when I wrote this when I realized there is a new version and it's still doing it (I was sure to kill and restart it).

tatokis, 2016-05-19:
Make sure the keyboard layout in your ckb settings matches your 
physical keyboard layout.

brettnem, 2016-05-19:
OMG
Thank you! That did it. Living without tildes was harder than I expected. :)

Apparently I had English (United Kingdom) set instead of English (United States)

closed 2016-05-19 by brettnem