ccMSC/ckb #356: M65 RGB control under OSX

realies, 2016-05-22:
I'm unable to find a way of controlling a M65 RGB mouse in the GUI under OSX. Any clues?

[image]

ccMSC, 2016-06-13:
Is this the M65 PRO? If so it's not supported yet, I'm going to be working on it soon.

realies, 2016-06-13:
It is the normal M65 RGB, tried it with even older M65 RGB and got the same screen.

realies, 2016-09-05:
Any ideas why the mouse is not showing up? I'm running OS X 10.11.6 and the device looks like this over usb:

`Bus 020 Device 023: ID 1b1c:1b12 CORSAIR MEMORY INC. Corsair M65 RGB Gaming Mouse`

schneems, 2016-10-19:
@ccMSC thanks for your work, just dropped you a few bucks. I'm on macOS Sierra and just bought a M65 Pro today. I didn't realize it wasn't working yet. Any idea on a timeline?