ccMSC/ckb #359: [Feature request] Specifying pauses in macros

IanDavey, 2016-05-26:
Suppose I want to switch to Chrome and open Gmail, and have a key combo Ctrl+Win+Alt+C assigned externally to do the first part.  To assign such a macro to G1, I might do:

```bash
echo macro g1:+lctrl,+lwin,+lalt,+c,-c,-lalt,-lwin,+t,-t,-lctrl,+g,-g,+m,-m,+enter,-enter > /dev/input/ckb1/cmd
```

The problem here is that the macro happens so fast either the OS application switcher or the external program handling the switch-to-Chrome shortcut can't keep up, and I end up with a Ctrl+T,G,M entered into my previous application.

Could there be a way to signal a brief pause in the macro command?  Such as the following (measured in ms):

```bash
echo macro g1:+lctrl,+lwin,+lalt,+c,-c,-lalt,-lwin,pause(500),+t,-t,-lctrl,+g,-g,+m,-m,+enter,-enter > /dev/input/ckb1/cmd
```

stephenhouser, 2016-08-13:
Would the work in this fork satisfy the delay feature you are looking for?

https://github.com/stephenhouser/ckb

IanDavey, 2016-08-13:
Yep - looking forward to it!

closed 2016-08-13 by IanDavey

stephenhouser, 2016-08-13:
I've still got to issue a pull request. Not *quite* closed yet :-)