ccMSC/ckb #360: k70 not working in Manjaro

ksjk1998, 2016-05-27:
I have done everything i can think of to get your driver to detect my keyboard. Nothing I did works. Perhaps I did something wrong, but everything your instructions told me to do, I did. this is the output of lsusb.

`Bus` 009 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub                                                                                                                              
Bus 005 Device 005: ID 1b1c:1b13 Corsair                                                                                                                                                    
Bus 005 Device 004: ID b58e:9e84 Blue Microphones Yeti Stereo Microphone
Bus 005 Device 001: ID 1d6b:0001 Linux Foundation 1.1 root hub
Bus 004 Device 001: ID 1d6b:0001 Linux Foundation 1.1 root hub
Bus 008 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 003 Device 001: ID 1d6b:0001 Linux Foundation 1.1 root hub
Bus 002 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 001 Device 003: ID 1b1c:0c06 Corsair 
Bus 001 Device 001: ID 1d6b:0001 Linux Foundation 1.1 root hub
Bus 007 Device 002: ID 2109:0812 VIA Labs, Inc. VL812 Hub
Bus 007 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 006 Device 004: ID 0bc2:5071 Seagate RSS LLC FreeAgent GoFlex Desk
Bus 006 Device 003: ID 04d9:fa51 Holtek Semiconductor, Inc. 
Bus 006 Device 002: ID 2109:2812 VIA Labs, Inc. VL812 Hub
Bus 006 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root `hub`

and lsusb -t if thats useful
``
/:  Bus 09.Port 1: Dev 1, Class=root_hub, Driver=ehci-pci/4p, 480M
/:  Bus 08.Port 1: Dev 1, Class=root_hub, Driver=ehci-pci/5p, 480M
/:  Bus 07.Port 1: Dev 1, Class=root_hub, Driver=xhci_hcd/4p, 5000M
    |__ Port 1: Dev 2, If 0, Class=Hub, Driver=hub/4p, 5000M
/:  Bus 06.Port 1: Dev 1, Class=root_hub, Driver=xhci_hcd/4p, 480M
    |__ Port 1: Dev 2, If 0, Class=Hub, Driver=hub/4p, 480M
    |__ Port 3: Dev 3, If 0, Class=Human Interface Device, Driver=usbhid, 12M
    |__ Port 3: Dev 3, If 1, Class=Human Interface Device, Driver=usbhid, 12M
    |__ Port 3: Dev 3, If 2, Class=Human Interface Device, Driver=usbhid, 12M
    |__ Port 4: Dev 4, If 0, Class=Mass Storage, Driver=usb-storage, 480M
/:  Bus 05.Port 1: Dev 1, Class=root_hub, Driver=ohci-pci/4p, 12M
    |__ Port 3: Dev 4, If 0, Class=Audio, Driver=snd-usb-audio, 12M
    |__ Port 3: Dev 4, If 1, Class=Audio, Driver=snd-usb-audio, 12M
    |__ Port 3: Dev 4, If 2, Class=Audio, Driver=snd-usb-audio, 12M
    |__ Port 3: Dev 4, If 3, Class=Human Interface Device, Driver=usbhid, 12M
    |__ Port 4: Dev 5, If 0, Class=Human Interface Device, Driver=usbhid, 12M
    |__ Port 4: Dev 5, If 1, Class=Human Interface Device, Driver=usbhid, 12M
    |__ Port 4: Dev 5, If 2, Class=Human Interface Device, Driver=, 12M
/:  Bus 04.Port 1: Dev 1, Class=root_hub, Driver=ohci-pci/2p, 12M
/:  Bus 03.Port 1: Dev 1, Class=root_hub, Driver=ohci-pci/5p, 12M
/:  Bus 02.Port 1: Dev 1, Class=root_hub, Driver=ehci-pci/5p, 480M
/:  Bus 01.Port 1: Dev 1, Class=root_hub, Driver=ohci-pci/5p, 12M
    |__ Port 5: Dev 3, If 0, Class=Human Interface Device, Driver=usbhid, 12M
``

tatokis, 2016-05-27:
Assuming you are on firmware 2.04, try compiling the fw2kb branch 
instead.

ksjk1998, 2016-05-27:
Whats the command for doing that If I may ask?

tatokis, 2016-05-27:
```
git clone https://github.com/ccMSC/ckb.git
cd ckb
git checkout fw2kb
./quickinstall
```

ksjk1998, 2016-05-27:
Ill see what this does. Thank you for sticking with me. I didn't expect someone replying in literally seconds after my post

closed 2016-05-27 by ksjk1998