ccMSC/ckb #364: Keyboard dies in ubuntu after firmware update in windows

Kira9204, 2016-05-29:
Hi!
I bought a K70 RGB today, plugged it in and installed the software on Ubuntu 16.04.
Everything worked perfectly fine until i later during the day updated the firmware in windows. 
Now the keyboard ignores all options and dies completely if i try to apply any animation.

tatokis, 2016-05-29:
Compile and install the fw2kb branch

Kira9204, 2016-05-29:
I removed and reinstalled with that branch but the issue persists.

Kira9204, 2016-05-30:
Everything works perfectly again!
Thank you!

ryanaltvater, 2016-05-31:
Same exact issue, and will try compiling that branch.

ryanaltvater, 2016-05-31:
I compiled the fw2kb branch, replaced the existing CKB.app, and I'm still experiencing the same issue. My keyboard is lighting up with the base colors of my profile but it's not triggering the animations and my keyboard is pretty much unresponsive. Pressing "asd" = "l;'" on my K70 RGB keyboard. This mess started happening after I updated the Windows firmware yesterday on my desktop (big mistake, as I use my MacBook and K70 for work).

ryanaltvater, 2016-05-31:
Also, I can't seem to load the demo profile or reset the keyboard either.

t3kg33k, 2016-06-07:
I've run into the same issue. Firmware updated within Windows, rebooted back into Ubuntu Mate 15.10, and my K65 keyboard no longer works. It looks as if the ckb-daemon crashes. I attempt to start the daemon, it runs for short time and then crashes again.

ryanaltvater, 2016-06-07:
@t3kg33k After contacting Corsair support, I ended up rolling the Windows CUE software (v1.15.36) and keyboard firmware (v1.30) back and it fixed everything for me.

t3kg33k, 2016-06-07:
@ryanaltvater Thanks. I'll try that tonight.
Corsair bastards need to include Linux drivers.

closed 2016-06-13 by ccMSC

ryanaltvater, 2016-06-13:
@ccMSC how come you closed this issue? Did you release a fix?

ccMSC, 2016-06-30:
Yes, v0.2.6 is compatible with the latest firmware.