ccMSC/ckb #365: mac

dreamcog, 2016-06-01:


tatokis, 2016-06-02:
Which K70 do you have? RGB or RGB Rapidfire?
And if so, what firmware version is it running?

If you have a K70 RGB and it runs firmware version 2.04, then you need to compile ckb from the fw2kb branch. Alternatively, someone has provided a package that you can try to install, in https://github.com/ccMSC/ckb/issues/298#issuecomment-218058262

If you have a K70 RGB Rapidfire, then unfortunately, it is not supported.

EmeraldGamer, 2016-06-02:
Wait rapid fire is not supported
will it be supported in the near future?

dreamcog, 2016-06-05:
thanks tatokis,
but it's doesn't work with my keyboard.

My keyboard is K70 RGB, and with firmware v2.04.
Use Mac OS X 10.11.5

I just uninstall the CKB, and the keyboard works, can type any letter as it before.
I am waiting for CKB could update for completely solve the problems:)

tatokis, 2016-06-05:
@dreamcog Install XCode and Qt as mentioned here:
https://github.com/ccMSC/ckb#building-from-source
then download the source code from https://github.com/ccMSC/ckb/archive/fw2kb.zip
and finally run quickinstall from it.

That should compile CKB with the fixes for firmware 2.04, if that doesn't work, then unfortunately I can't help as I do not have a Mac.

dreamcog, 2016-06-05:
@tatokis Thanks a lot tatokis ,I will try it tomorrow, Now it's very late in china:)
hope it works. I will reply to you after i try.thanks

tatokis, 2016-06-05:
No worries, I forgot about it completely, but there is another build as 
mentioned in 
https://github.com/ccMSC/ckb/issues/298#issuecomment-223824964 , so you 
might want to try that one first.

dreamcog, 2016-06-06:
@tatokis I tried it this morning.and still a problem here : [https://github.com/ccMSC/ckb/issues/298#issuecomment-223824964](url)

dreamcog, 2016-06-06:
@tatokis I tried it this morning.and still a problem here : https://github.com/ccMSC/ckb/issues/298#issuecomment-223824964

mhennessie, 2016-06-08:
I had to build my own for the K65 RGB on OS X running 2.0.4. The one pre compiled [here](https://github.com/ccMSC/ckb/issues/298#issuecomment-216055305) by @4t0m1k didn't work for me either. After installing QT I was able to do it in a few mins.

closed 2016-06-13 by ccMSC