ccMSC/ckb #366: OSX

EmeraldGamer, 2016-06-02:
I finally install the program and then it says no devices connected and im confuesed pls help i need this to work. I didn't spend 170 dollers for this BA keybaord just to have it not work!!!

EmeraldGamer, 2016-06-02:
Never mind that 
there is a flashing light on my keyboard (K70 RGB Rapid fire) And the light looks like a download icon do i need to install something for the program to work? please let me know what i need to do

EmeraldGamer, 2016-06-02:
Maybe someone could release a full mac OS X start up and install vid?

EmeraldGamer, 2016-06-02:
also when i change my keybaord to a type mode it says no devices connected

tatokis, 2016-06-02:
The K70 RGB Rapidfire is NOT supported, only the original K70 RGB is.
See #319

EmeraldGamer, 2016-06-02:
Will it be supported in the near future??? :'(

EmeraldGamer, 2016-06-02:
i really love this keyboard and i want the RGB to work

Kira9204, 2016-06-02:
EmeraldGamer: I feel that you have a somewhat demanding, negative tone here. The "pls help i need this to work. I didn't spend 170 dollers for this BA keybaord just to have it not work!!!" is, at least for me, out of line.
The developers of this software does not get paid for their work, they do it in their spare time.
As such, do not make demands but rather offer them your help in anyway you can.
As you can clearly see there is a list of supported hardware. Your hardware is not on that list.
Politely ask if there will be support next time. And post in single posts, not 10 multiple small ones.
These are personal opinions i felt the need to express.
Depending on how new your keyboard is, you should have the option of returning it to another supported model.

ccMSC, 2016-06-13:
I've unfortunately not had any time to work on this project lately, however I do intend to add support for the newer models. There are several of them so I'm going to be looking at them one by one over the next week.

NotSideEffect, 2016-06-13:
Looks like Kira has very thin skin if they get upset over someone asking for help like this. If you see people like him in a negative way then you shouldn't even be here because there are a lot of people like this. They don't bother me at all because I would be here to help not to judge. I am only going to judge you because 99% of your post was you crying over text and not offering any help.

ccMSC, 2016-06-18:
I've added experimental support in a new testing branch. There were a lot of threads started over this so I'm closing all of them and moving the discussion there: #386

closed 2016-06-18 by ccMSC