ccMSC/ckb #370: ckb not working on firmware 2.04

batchpert, 2016-06-08:
Hi I bought a new Strafe RGB keyboard and then connected it to windows to update it before connecting it to my mac. When i first connected it the standard red keys and white WASD and directional keys appeared. I then launched version 0.2.5 of ckb after this the keyboard appeared on the ckb screen but when i cycled through the modes the lights on the keyboard did not change. Additionally now that ckb was active it was not registering any key presses making my keyboard technically useless.

I have asked a friend who uses this and he said the only thing he had that was different is he is still on firmware 1.33 and doesn't know if this has an effect on the ckb software. I don't know if it is possible to downgrade to that firmware version or how to do that. 
Any help appreciated, thanks.

batchpert, 2016-06-09:
using #347 I managed to get it working perfectly

jvosk, 2016-06-12:
Same problem as you, fixed in the same manner (compiling from branch fw2kb) -- seems this issue would be easily resolved if the README clarified in the Troubleshooting or Known Issues sections that the fw2kb branch is required to be compatible with the Corsair 2.04 firmware (if it's not possible to have a more elegant solution, even as 'simple' as selective compiling given a command line option). Cheers!

closed 2016-06-13 by ccMSC