ccMSC/ckb #372: Keyboard detection issue

stevehaber1, 2016-06-09:
Tried the master branch as well as FW2KB build from source - neither are getting the application to detect the hardware.

Brand new keyboard - only things done to it were to install the latest firmware on my PC, and to write a profile to the device memory before I even found out about CKB. 

Currently on OS X 10.11.5

akatrevorjay, 2016-06-11:
I have this issue as well, and I *also* have a brand new G70 LUX RGB. It must be related.

akatrevorjay, 2016-06-11:
Corsair Gaming K70 LUX RGB Keyboard :

              Product ID: 0x1b33
              Vendor ID: 0x1b1c  (CORSAIR MEMORY INC.)
              Version: 1.05
              Speed: Up to 12 Mb/sec
              Manufacturer: Corsair
              Location ID: 0x40110000 / 4
              Current Available (mA): 1000
              Current Required (mA): 500
              Extra Operating Current (mA): 0

kayji, 2016-06-11:
I also see the same "error" message: "No devices connected". Even though, the keyboard is working. I can use all the function keys as well. But I can't change keys using ckb.

stevehaber1, 2016-06-11:
@akatrevorjay @kayji - Unless you performed the same firmware update, I think it may be unrelated. In any event, I'm not sure what you tried but I was able to fully resolve my issue. There was a post in issue #240 to the pkg version of one of the feature branches that fully fixed this for me.

1. Remove the CKB app from applications
2. Restart
3. Download / install the feature branch pkg - https://github.com/ccMSC/ckb/issues/298#issuecomment-223824964

Give it a shot and let me know how it turns out.

akatrevorjay, 2016-06-11:
Oh, I did update it to 2.x, that lsusb was from before I had.

akatrevorjay, 2016-06-11:
I had to add my device IDs to the codebase to get ckb to detect my usbid, I'll keep playing with it, Currently using fw2kb branch.

tatokis, 2016-06-11:
According to README.md the Lux is not supported. 
There's an issue already open about this: #320

stevehaber1, 2016-06-11:
In any event, closing this one. Please raise new for yours if you wish to track.

closed 2016-06-11 by stevehaber1