ccMSC/ckb #373: ./quickinstall: line 200: 19598 Aborted

haydenjames, 2016-06-11:
Fresh install of:
Kubuntu 16.04 LTS

With:
K65 lux

Quick install gives this error:
`Start ckb now [Y/n]? Y
./quickinstall: line 200: 19598 Aborted                 (core dumped) $APP --close > /dev/null 2>&1`

tatokis, 2016-06-11:
That happens because the daemon crashes as your keyboard is not supported.

haydenjames, 2016-06-12:
Ahh! K65 **LUX** RGB not supported you are saying? I will exchange it at BestBuy for the regular "K65 RGB".

tatokis, 2016-06-12:
Yeah, see https://github.com/ccMSC/ckb/issues/320
Also, might want to try the fw2kb branch, as it contains fixes for the new firmware (2.04).

tatokis, 2016-06-12:
However, since there was a crash, please try running ckb with the K65 unplugged, to see if it will start successfully or not.
If it doesn't start, there is chance that you came across another issue as well.

haydenjames, 2016-06-13:
Thank! I picked up the regular k65. This works perfectly!

closed 2016-06-13 by haydenjames