ccMSC/ckb #375: Shutdown Hang

Da-Boom, 2016-06-12:
Ok. so i have an issue (obviously)

So i have a Corsair STRAFE RGB running the latest firmware (2.04)..
I am using the latest fw2kb driver with Ubuntu 16.04

And I also have the usbhid.quirks in the grub CMDLINE

My problem is that when i shut my PC down, it hangs with a usbhid error of -110.. 
as far as i can tell this shouldn't happen, and it's really annoying as sometimes it will hang for 30 secs, sometimes it will do so indefinitely until i force the PC to turn off with the power button

I did notice this error also occurring in my DMESG after booting (it seems it's just before Ckb starts up) so i have linked a pastebin of the offending parts

[http://pastebin.com/w0wvYdy6](http://pastebin.com/w0wvYdy6)

If you need anything else, dont hesitate to ask :)

ccMSC, 2016-06-13:
Does it happen even with the driver disabled? If so then I'm not sure there's anything I could do about it.

Alkeryn, 2018-03-07:
@ccMSC Have same issue
though if i kill the ckb-daemon (using systemctl stop ckb-daemon) prior to powering off
it poweroff quickly
still the daemon need a while to shut down even through "sytemctl stop"
but once it is dead i shutdown normaly
so yeh
btw sorry for the necrobump but i end up here while searching on google about that issue

OvermindDL1, 2018-03-07:
@Alkeryn For note, all current ckb development is done at https://github.com/ckb-next/ckb-next as this repo is currently dead.  :-)