ccMSC/ckb #376: Corsair K65 LUX RGB not working under 10.10.5

TonyK54, 2016-06-12:
I should clarify. It works. OS X detects the keyboard by name and I can switch the modifier keys.

But ckb does not work. Nothing tried works. I've searched with Google and have tried suggestions here and nada. 

Have tried on different ports, on a USB hub and even on a new MacBook Pro with 10.11. At each turn cbk responds with device not detected. 

Tried to update the plist file and add the logging values. No log files created. 

All I want to do for my legally blind wife is set a different color for the F and J keys, the CRTL keys and the ESC key. The other colors can stay at the default RED color (except for the group on the left that are white. they need to be red too).

Thoughts on why 0.25 does not work. Tried 0.22 and still did not work.

Yes, I've completely removed ckb from the system before re-installing. Multiple reboots. Still nada.

Thanks,

tatokis, 2016-06-12:
Your keyboard is not supported, #320

TonyK54, 2016-06-12:
Bummer.

closed 2016-06-12 by TonyK54