ccMSC/ckb #377: CKB fails to detect Corsair Sabre RGB/No devices connected

nbulischeck, 2016-06-12:
Linux box 4.6.2-1-ARCH #1 SMP PREEMPT Wed Jun 8 08:40:59 CEST 2016 x86_64 GNU/Linux

I can use the device just fine, but ckb never picks it up. Forever "No devices connected". I've tried every conceivable combination of the quirks, even the 0x4 suggestion and nothing worked. The CKB daemon is 100% enabled and running.

Terminal output is: hid-generic 0003:1B1C:1B2F.0002: usb_submit_urb(ctrl) failed: -1

ccMSC, 2016-06-13:
Should be fixed by the new version release. Issue with the new firmware.

nbulischeck, 2016-06-13:
New version release of ckb? Or of the firmware?

ccMSC, 2016-06-13:
new version of ckb: https://github.com/ccMSC/ckb/releases/tag/v0.2.6

nbulischeck, 2016-06-13:
Still no luck getting it to recognize the device.

SethBarberee, 2016-06-14:
I have a Strafe RGB with the newest firmware and it is detected properly. I kept the USB quirks in my kernel to avoid that error above. In the end, it works great for me!

nbulischeck, 2016-06-15:
Not sure what to tell you because it still doesn't work for me =/

ccMSC, 2016-06-18:
You probably have the newer sabre model. Try compiling this branch and see if it works: https://github.com/ccMSC/ckb/tree/newdev

If not, please see #311

closed 2016-06-18 by ccMSC

nbulischeck, 2016-06-18:
It worked! Thanks for your help.

sabrehelp, 2017-02-23:
Hi, I am now having this same issue. Sorry if its a silly question but how do I "Compile" the above branch? 
Thanks for your time!

frickler24, 2017-02-23:
@sabrehelp please have a look at the more recent fork by [mattanger/ckb-next](https://github.com/mattanger/ckb-next).

sabrehelp, 2017-02-25:
I keep getting a "you cannot open the application because it is damaged or incomplete" after quick install. @frickler24