ccMSC/ckb #378: CKB 0.2.6 No device Connected K70 RGB LUX OS X El Capitan

Tapdamonkey, 2016-06-14:
CKB gives me a no device connected when i run the program.  I updated my firmware on the K70 RGB to latest, works perfectly on windows.  I downloaded the package to install my drivers, i have tried re-install, changing the poling rate, and rebooting the laptop. Have not done any modifications to source code as i am not too familiar with coding, any help would be appreciated, thanks!

tatokis, 2016-06-14:
The Lux is not supported yet.

Tapdamonkey, 2016-06-14:
Crap, guess ill wait it out then,  Do you know the difference between the LUX and Standard RGB?

esparkman, 2016-06-14:
I was curious on the LUX support as well.

ccMSC, 2016-06-18:
Hi, there have been a lot of threads about new devices and I don't want to divide attention between all of them. I've opened up a new thread for them all: #386 - Please continue discussion there, thanks!

closed 2016-06-18 by ccMSC