ccMSC/ckb #380: Can't Get Ckb Working on Linux Deepin

dabfalcon, 2016-06-15:
Linux Deepin is Debian-based nowdays. I can install Ckb on the OS from GitHub but can't get it to actually change any colours. What can I do to help troubleshoot this issue?

ccMSC, 2016-06-18:
What specifically is the problem? Does the GUI show the keyboard but you're unable to change anything, or are you getting the `Driver inactive` or `No devices connected` screen?

dabfalcon, 2016-06-22:
The former happens. I believe the keyboard even stopped working, as well. I'm now on another computer using Deepin and facing the same problem. Its probably trouble with dependencies. apt-cache search qt5 gives like 1 million results but apt-get install qt5 doesn't work. What should I try?

tatokis, 2016-06-22:
Considering you already got ckb compiled, I don't think it is a Qt issue, as that only affects the GUI and not the daemon.

Have you added the correct quirk parameters to GRUB?

Which device do you have?
Some of the new devices require code which exists on the "newdev" branch.

dabfalcon, 2016-06-25:
Issue resolved. For some reason I'm worked on this install. Maybe Ckb caught up to the firmware I was running? In any-case, re-ran quickinstall fixing errors by downloading appropriate packages. USB Quirk and is all good. Thanks

ccMSC, 2016-06-30:
It's possible; the latest firmware was incompatible with the master branch for quite a while. Glad you got it working though.

closed 2016-06-30 by ccMSC