ccMSC/ckb #386: MEGATHREAD: Newer keyboards/mice (K65 Lux, K70 Rapidfire, M65 Pro)

ccMSC, 2016-06-18:
I've added support for the K65 Lux and K70 Rapidfire in this branch: https://github.com/ccMSC/ckb/tree/newdev

It does *not* support the M65 Pro as nobody has told me the USB device ID for it yet.
* To find the ID if you're on Linux: run `lsusb -d 1b1c:` in a terminal and post the output here
* To find the ID if you're on OS X: Use "About This Mac" and then click the "More Info" button. You'll see a list of devices on your system and you should be able to find the mouse under your USB devices. Click on it and you'll see some detailed information in the page below - take a screenshot of that page.

A few of you tried to get the keyboards working by editing the device IDs but found that the lighting didn't work - I think I know why. It should be fixed in this version.

Hopefully this version works for everyone and I can release it soon. Let me know which device you have and whether or not the test version is working. I can also provide an OS X binary package if anyone needs it.

rbieb, 2016-06-18:
Does the support for the rapidfire k70 include support for the Corsair K65 RGB Rapidfire? If not, please let me know so I can get you the info you need.

gtjoseph, 2016-06-18:
K65 Lux  1b1c:1b37 and Scimitar 1b1c:1b1e working fine in the newdev branch.

THANKS!!!

jashephe, 2016-06-19:
Thanks for all of your hard work! I built the newdev branch, but the K65 Rapidfire doesn't seem to be supported yet. Not sure if you have this information already, but here's the device ID for the K65 Rapidfire:
[image]

tatokis, 2016-06-19:
@rbieb @jashephe please try applying the following patch and then compiling ckb again.
https://vps.tasossah.com/ckb/K65_Rapidfire.patch

rbieb, 2016-06-19:
@tatokis 
Works beautifully, thanks a lot!

jashephe, 2016-06-19:
@tatokis I still get "No devices connected", even for the patched version.

tatokis, 2016-06-19:
@jashephe Make sure the daemon is restarted. Close the ckb application and run
```
sudo killall -9 ckb-daemon && sudo ckb-daemon
```
and then start the ckb application again.

If it still doesn't recognise it please post the output of the above command.

Alternatively, try rebooting.

jashephe, 2016-06-19:
Sure, I had killed the daemon before, but here's the output of killing it and starting it manually:

```
ckb: Corsair RGB driver beta-v0.2.6+d01
[I] Root controller ready at /var/run/ckb0
[W] iterate_devices_usb (usb_mac.c:629): Unable to seize USB handle, continuing anyway...
[I] Connecting  at /var/run/ckb1
[W] iterate_devices_hid (usb_mac.c:751): Failed to open device: e00002c5
[W] iterate_devices_hid (usb_mac.c:751): Failed to open device: e00002c5`
```

tatokis, 2016-06-19:
Since it worked for rbieb who runs linux, it is probably a Mac-specific issue.

However, the only modification that needs to be done specifically for Macs is on line 773 on usb_mac.c, to include "P_K65_RFIRE", which the patch already does.
So assuming the patch hasn't failed, I am not sure what is going on.

I also run Linux, and don't have any knowledge of how Macs handle USB devices, so unfortunately I can't help you.
Will have to wait for @ccMSC

ajmicheli, 2016-06-19:
K70 Rapidfire (non RGB):
Bus 003 Device 002: ID 1b1c:1b36 Corsair

lukeberry99, 2016-06-19:
Out of town until next Friday, if no one has gotten back to you by then I'll be able to update you with the M65 Pro RGB device ID.

kirishima262, 2016-06-20:
I purchased a k65 lux for my mac (10.9) and was disappointed to find that ckb didn't detect it but after building this latest dev version (and restarting) it works!!!! thank you so much for this wonderful software. Also your build system is really simple; it's often a challenge to build projects like this but one script and you're done ^^

Corsair should hire you as a dev. it's really poor that they have no native support for osx/linux.

Kithop, 2016-06-20:
I just compiled and ran the updated newdev branch version on my Linux machine - success!  K70 RGB Rapidfire is now working with animations and such.

TekNicholas, 2016-06-20:
M65 RGB PRO
$ lsusb -d 1b1c:
Bus 004 Device 004: ID 1b1c:1b13 Corsair 

Let me know if you need anything more. Would love to have this added.

superak, 2016-06-23:
I just got a k65 lux for my mc (10.10) and installed the latest version CKB (v0.2.6). I am pretty new to the scripts and code stuffs, so I used the simplest way to install CKB (download ckb.pkg and run it). 
Even I have tried all the usb port and external hubs, multiple restarting, it cannot detect the keyboard shows message "no devices connected". Any advice will be appreciated...
PS: the keyboard just works fine in windows system.

tatokis, 2016-06-23:
You need to compile ckb using the quickinstall script.

There are instructions in the readme ( https://github.com/ccMSC/ckb/tree/newdev#building-from-source ), but make sure to get ckb from this link instead, which contains support for the K65 Lux:
https://github.com/ccMSC/ckb/archive/newdev.zip

superak, 2016-06-25:
@tatokis Thanks a lot. I installed gt5 then used the quickinstall script, and the keyboard works perfect. Thumb up...

Neldogz, 2016-06-26:
Thank you for bringing this to Linux

M65 RGB PRO
$ lsusb -d 1b1c:
Bus 001 Device 005: ID 1b1c:1b2e Corsair

Please let me know if there is any other information that you need.

skiller23, 2016-06-28:
I just got the Corsair RGB Lux k70 and using this ckb software won't recognize the keyboard. I've been reading this thread and have come to the conclusion that the software is not optimized for this new keyboard. Any way to get it to to work?

jashephe, 2016-06-28:
@tatokis Just wanted to report that I had an opportunity to try this again and was able to get a K65 Rapidfire working on a mac using the `newdev` branch with your patch. It seemed like unplugging and re-plugging the keyboard while ckb was open seemed to do the trick (although I can't be certain that was all there was to it, since I also did a fresh build and patch).

Gausgovy, 2016-06-28:
I was able to get the newdev build to recognize the k70 LUX RGB But changing the lighting would not change anything. Shortly after I got it to recognize the keyboard stopped typing. I fixed it with a simple reboot then deleting all CKB files and unplugging the keyboard once. What I did to get it to recognize was install with the terminal command, then just fiddling around being dumb in the files, I didn't think it would do anything, but I opened package and preinstall then unplugged my keyboard and plugged it back in then suddenly it recognized it. When I tried changing the lighting it just turned the lights off then back on. I most likely will not try this for a while, the lights just bother me because the rest of my setup is blue and it has red lights. I will check this page daily to see if anybody has a fix though. It may because k70 lux rgb isn't supported, but when I checked the product ID matched the one provided.

newnewcorsair, 2016-06-28:
Hi,  Ive tried to patch for the K65_Rapidfire.patch but keep getting this error?

can't find file to patch at input line 5
Perhaps you used the wrong -p or --strip option?
The text leading up to this was:
--------------------------
|diff --git a/src/ckb-daemon/usb.c b/src/ckb-daemon/usb.c
|index c9af78c..f529891 100644
|--- a/src/ckb-daemon/usb.c
|+++ b/src/ckb-daemon/usb.c
--------------------------

I have downloaded the patch and moved it into the "contents" folder of the CKB app files? then i open terminal, and switch to the contents folder, and try running the patch?  any ideas?

CommandMasterGeek, 2016-06-29:
I just bought these two latest models of mouse and KB and could not get either to work on OSX 10.11.5. As requested I have uploaded the "About this Mac" info' on both devices. It would be great if you could add them. I'm not a programmer so a new .pkg to download would be awesome. Thanks for the great work. Its much appreciated.

[image]
[image]

taisph, 2016-06-30:
From Ubuntu 16.04:
```
[94622.893433] usb 3-9.4.2: New USB device found, idVendor=1b1c, idProduct=1b2e
[94622.893437] usb 3-9.4.2: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[94622.893438] usb 3-9.4.2: Product: Corsair Gaming M65 Pro RGB Mouse
[94622.893439] usb 3-9.4.2: Manufacturer: Corsair
```

taisph, 2016-06-30:
Attempted to use the M65 code to support M65 Pro in my fork here: https://github.com/taisph/ckb ... seems to timeout on retrieving the mouse profile though.

ccMSC, 2016-06-30:
All the new devices should be added now. Thanks everyone.

@taisph that's a pretty common issue, as long as the mouse itself works I wouldn't be bothered by it.

CommandMasterGeek, 2016-07-02:
Thanks so much. I followed you OSX build instructions for this fork and they worked perfectly. My new K70 RGB Rapidfire KB and M65 Pro RGB mouse were recognised first go with the new build on  OSX 10.11.5 on my iMac. Now having fun trying out the S/W and options. Will advise of any bugs. Great job!

robgc, 2016-07-02:
Hi! Here you have the K70 LUX RGB's deviceID:

`Bus 003 Device 002: ID 1b1c:1b33 Corsair`

Thanks a lot for your amazing work!

Gausgovy, 2016-07-02:
so I got it to recognize my k70 LUX RGB, but whenever it does the lights turn off and the keyboard basically disconnects from my computer, then it reconnects and works again

koanya, 2016-07-02:
I have a K70 rapidfire (non-rgb) and it not recognize. I compiled the New-dev branch. It's the ID "Bus 003 Device 002: ID 1b1c:1b3a Corsair". It's strange that I havn't the same ID than @ajmicheli.

Another problem (or maybe not). When I launch ckb-daemon I have only : 
juil. 02 21:18:04 Archlinux systemd[1]: Started Corsair Keyboard Daemon.
juil. 02 21:18:04 Archlinux ckb-daemon[472]:     ckb: Corsair RGB driver beta-v0.2.6+d01
juil. 02 21:18:04 Archlinux ckb-daemon[472]: [I] Root controller ready at /dev/input/ckb0
I don't know if it's normal that I don't have a line "keyboard not recognizeé" or something like that. And during the boot, I have the message "usbhid: 9-2:1.1: couldn't find an input interrupt endpoint". Maybe it's because this message than my keyboard is not recognize.

In all case, thank you for your hard work on this driver :)

Gausgovy, 2016-07-03:
Update on k70 Lux RGB. It is recognizing it as a standard k70, the program attempts to change the lights and does nothing.

ganoz, 2016-07-04:
K70 Rapidfire (non RGB) not working:
[image]

Thanks a lot for your amazing work! :)

bladejlm, 2016-07-06:
Thanks in advance for your great Work ... I have K70 Rapidfire (spanish edition) that is the "About my mac" usb information:
[image]

I'd like a refresh of your driver for mac to get full funcionality on osX

Thanks!

koanya, 2016-07-12:
If someone want to use K70 rapidfire non-rgb, you just need to modify the file usb.h in src/ckb-daemon/ in the newdev branch. Replace the line "#define P_K70_RFIRE     0x1b38" by "#define P_K70_RFIRE     0x1b3a" and the line "#define P_K70_RFIRE_STR "1b38"" by "#define P_K70_RFIRE_STR "1b3a"" (1b3a is the id of my k70 rapidfire. You can check with the command "lsusb -d 1b1c:"). Then just build the program. Thanks @piyatat for the info.

horen20wash, 2016-07-12:
Cab from the needed branch works fine for K70 RGB LUX

Gausgovy, 2016-07-15:
Horen are you on mac or linux?

horen20wash, 2016-07-15:
I am on Mac.

Thx,
Horen

lukeberry99, 2016-07-15:
I'm currently not in a position to compile the dev branch, is there any ETA on when this will be released or is anyone kind enough to package this up for Mac and send it to me? If not, no worries

chrisengelsma, 2016-07-31:
Corsair K65 Lux RGB
Mac OS X 10.11.6

Very excited about getting this to work, however after following the instructions on the OSX install, the ckb.app fails to launch. Launching from a terminal window will output this:

`LSOpenURLsWithRole() failed with error -10810 for the file /Applications/ckb.app.`

Any thoughts?

bwoodruff, 2016-08-03:
[image]

K70 non-RGB non-Rapidfire LUX

bwoodruff, 2016-08-03:
Was able to get the above working by editing the usb.h file as @koanya suggested.

amnolan, 2016-08-07:
Followed the instructions on building locally using Xcode and QT with the latest branch. I was able to get the K65 Lux working with MacOSX. Thanks.

poppeach, 2016-08-09:
[image]

K70 Non-RGB LUX

mapintar, 2016-08-12:
I have a k70 rapidfire ABNT2 (Brazilian portuguese) and replace the line "#define P_K70_RFIRE 0x1b38" by "#define P_K70_RFIRE 0x1b3a" and the line "#define P_K70_RFIRE_STR "1b38"" by "#define P_K70_RFIRE_STR "1b3a"". Keyboard now is detected by ckb-daemon, but ckb not have suport for ABNT2.

mrgaolei, 2016-08-17:
Corsair Gaming K65 RGB RAPIDFIRE Keyboard ：

  产品 ID：	0x1b39
  厂商 ID：	0x1b1c  (CORSAIR MEMORY INC.)
  版本：	2.05
  序列号：	0A04101AAEAA10025504DC66F5001943
  速度：	最大 12 Mb/秒
  制造商：	Corsair
  位置 ID：	0x14300000 / 6
  可用电流 (mA)：	500
  所需电流 (mA)：	500
  额外的操作电流 (mA)：	0

bwoodruff, 2016-08-24:
Is a new binary with the above mentioned changes planned? Is there anything more we can do to help @ccMSC?

elXaptation, 2016-08-24:
Hi,
I have the K70 RGB Rapidfire

blah:~$ lsusb -d 1b1c:
Bus 001 Device 005: ID 1b1c:1b38 Corsair 

The ckb-newdev/src/ckb-daemon/usb.h file lists this id as 
blah:~$ cat Downloads/ckb-newdev/src/ckb-daemon/usb.h | grep 1b38
#define P_K70_RFIRE     0x1b38
#define P_K70_RFIRE_STR "1b38"

ckb control panel recognizes the device as connected, but can not control it.  No light control. I only get the NumLck, CapLck , OtherLED flashing.

I don't have a windows machine to update firmware. Could this be the problem. I guess the current ckb-newdev branch build is looking for K70 to have the newest firmware? Please confirm if possible. 

Any other ideas?

Thank you.

bwoodruff, 2016-08-24:
Hi @elXaptation 

The newest firmware was required in my experience. Without it ckb did not even "see" my keyboard. Updating the firmware from a Windows PC then connecting it to ckb worked perfectly (after making the above mentioned required changes).

Ben

boatchum, 2016-08-27:
For some reason I cannot install QT5. and they will not offer further support to me. I have even tried on all the 3 mirrors provided from qt. Is there a workaround QT5? 
Keyboard is recognized, CKB is installed (it just doesn't show anything other than settings when app is open), and showing the default Red/White LEDs. So basically the install begins and ends with this error every time: Could not download archive: http://download.qt.io/online/qtsdkrepository/mac_x64/desktop/tools_qtcreator/qt.tools.qtcreator/4.1.0-0qtcreator.7z : Connection closed

Please help thanks.

fgblomqvist, 2016-09-03:
@bwoodruff I got the same keyboard as you (same ID) but it's dead when I run CKB (I created a separate issue here: https://github.com/ccMSC/ckb/issues/453). What firmware version are you on? Which branch/revision of CKB are you using?

bwoodruff, 2016-09-03:
Hey @fgblomqvist -- I'm using FW 2.04 (if you haven't updated from a Windows box that could definitely be a problem). I built ckb from the newdev branch after making the changes I mentioned above.

fgblomqvist, 2016-09-03:
@bwoodruff Didn't realize I had to specifically replace the K70 rapidfire one, once I did and recompiled I got it to work! Thanks a lot.

So to clarify: If you have K70 LUX (no RGB, no Rapidfire) you have to edit the K70 Rapidfire section of usb.h so that it looks like this:
```
#define P_K70_RFIRE     0x1b36
#define P_K70_RFIRE_STR "1b36"
```

elXaptation, 2016-09-06:
>     <reply+012ed9c6f5887bc3a71df74720a5dfb15208249fe2e3fd7292cf0000000113d596b
>     K70 Rapidfire, M65 Pro) (#386)

Hi Ben,

 	Just closing the loop.

 	Upgrading the firmware using a Windoze machine solved my problem. 
Now the keyboard is recognized and ckb is controlling it properly.

 	Thank you.

 	And thank you CKB contributors.

richardspiers, 2016-09-08:
Just confirming that my K70 Lux RGB is working fine under the newdev branch code. Worked as plain keyboard without being detected by ckb under the old master branch.

rkingsbury, 2016-09-21:
Can anyone with a K70 non-RGB LUX confirm that Ubuntu 16.04 recognizes it as a standard keyboard? Is it necessary to enable the USB quirks? I've done so, and updated FW to 2.04, but my keyboard is unusable.

I'd really like it to work with the regular USB HID driver before I start messing with ckb...

fgblomqvist, 2016-09-21:
@rkingsbury Yes it does. At first Ubuntu did not recognize it, but after updating the firmware in Windows to 2.04 Ubuntu started to recognize it as well, without CKB installed. I did not enable any USB quirks as far as I remember.

kd8nzg, 2016-09-22:
I am so confused, how do I get my k70 rapidfire to work on MacOS?

horen20wash, 2016-09-22:
Please checkout the source code
Switch to newdev branch
Install Qt
Follow the description in Build.md for Mac.
Your keyboard will work fine.

kd8nzg, 2016-09-22:
I built it but it says no devices connected...any solutions? This is all new to me, thank you for your patience

bwoodruff, 2016-09-22:
@kd8nzg,

What version of Mac OS are you running? Have you upgraded to macOS Sierra?

[64 more replies]