ccMSC/ckb #388: [Feature request]

rbieb, 2016-06-19:
It would be nice if we were able to change the key at which each animation starts.

For example, the starting point for the ripple effect is the i-key, would be great if it could be changed to something else, too.

tatokis, 2016-06-19:
https://vps.tasossah.com/ckb/ckb-ripple_offset.patch
Is this what you mean?

rbieb, 2016-06-19:
I'd have to take a closer look at how exactly this works, but I think it it. how exactly would I use that once I've changed it?

tatokis, 2016-06-19:
It adds two new options in the Ripple animation settings, which let you change the X and Y offset of the center.
[image]

rbieb, 2016-06-19:
Ah, I see. But what I meant was this: How do I actually apply that patch file?

tatokis, 2016-06-19:
Download it to the ckb directory that already has the patch for your keyboard, run 
```
patch -p0 < ckb-ripple_offset.patch
```
and recompile ckb.

rbieb, 2016-06-19:
Thanks, I get this error though:

    patching file src/ckb-ripple/main.c
    patch unexpectedly ends in middle of line
    patch unexpectedly ends in middle of line

tatokis, 2016-06-19:
Ignore the errors, should work fine

rbieb, 2016-06-19:
Got it, works beautifully, thank you!

batchpert, 2016-06-19:
I like the idea of this but how do you do the

> Download it to the ckb directory that already has the patch for your keyboard, run

patch -p1 < ckb-ripple_offset.patch

and recompile ckb.

Sorry if its something simple i am not the best at this sort of stuff but would really like to get this effect to work without messing something up

batchpert, 2016-06-21:
I just wanted to add something to this but currently it seems to be set to (unless ive missed something) that if i was to have a ripple going across a line of keys say for example the F keys but i only wanted the ripple to go off when i pressed the pause key the ripple is only seen on all the keys its set to.
However this would mean that if i wanted the ripple to be seen across all the F keys then i would need to set all those keys to function in the same manner so i could press any of the F keys and they would all create the same ripple?

Am i right in thinking this and if so is there some way to change this so i could for example press the pause key and a ripple is shown somewhere else?

ccMSC, 2016-06-30:
You're correct, there's no way to separate the trigger keys from the animation keys. I've considered adding it but there are more important features I want to get around to first.