ccMSC/ckb #392: readlink: illegal option -- f in branch:newdev

mrgaolei, 2016-06-20:
In `newdev` branch, `./quickinstall`:
```
readlink: illegal option -- f
usage: readlink [-n] [file ...]
Preparing build files...
No suitable qmake found.
Please make sure you have Qt5 installed.
Re-run this program with QMAKE=/path/to/qmake to select qmake manually.

Press enter to exit.
```

tatokis, 2016-06-20:
Do you have Qt5 installed? If so what version?
Did you follow the instructions from https://github.com/ccMSC/ckb#building-from-source ?
Please provide more info.

ccMSC, 2016-06-30:
The readlink thing is actually a harmless error, though I should probably do something to suppress it.

If you installed Qt5 via brew, you might want to check out this patch: https://github.com/ccMSC/ckb/pull/324