ccMSC/ckb #394: OSX SCIMITAR LIGHTING

OMGojira, 2016-06-20:
just got the scimitar today, and immediately got ckb. love it so far i have both windows and mac. so far i have to say ckb is almost better in a lot of ways. It seems easier to navigate/edit.

BUT... i wouldnt be posting this in the issues if i didnt come across one issue that is upsetting me pretty badly.

when trying to create a new profile and change the color of the lights, the light by the numberpad will not change color. Some of the other built in profiles i am able to change and alter slightly but others have the same issue. it seems to be just the one light also....

i have kind of found a way around the device not connected issue by just closing the application unplugging the device starting the application and then plugging the device back in.

THIS ONE LIGHT THOUGH!!!!!!!!!!

PLEASE HELLP!!!

os x yosemite 10.10.5
macbook pro 15-inch retina

let me know if there is anything else i could possibly give you to help resolve. At this point if i could get this to work i would PAYYY i really like the feel of this mouse.

OMGojira, 2016-06-20:
i have kind of figured it out. but for some reason i cant get the animation and actual click to work at the same time now

OMGojira, 2016-06-20:
nevermind it broke again. now the profiles i was able to customize lasted about 30 minutes before that one light changed to a random yellow.

OMGojira, 2016-06-20:
ok the random colors were from the dpi. still trying to figure out how to animate the colors on click while keeping the click working. This could be simple but i may just be overlooking this. let me know thanks.

ccMSC, 2016-06-30:
There's not currently any way to trigger animations via click. In v0.3 it should be possible to do what you're looking for.

OMGojira, 2016-07-02:
Perfect thanks :) I love it so far it's got it's quirks but nothing crucial that I've noticed thanks again!!

mrnoisytiger, 2016-08-07:
If you want to set a custom color for that DPI light, so that the colors are not tied to a specific DPI, go to the performance tab and uncheck "Use DPI Indicator." That lets you set a custom color as you would any other light, as well as do animations on that light.

closed 2024-09-04 by Maefreric