ccMSC/ckb #398: GUI Doesn't look right...

adamleblanc501, 2016-06-25:
So when I installed this onto my Arch Linux everything seemed to install correctly, but when I open the program it shows nothing like the image that is shown on the readme in github.

Here is what it looks like: 
[image]

Any solution to this?

tatokis, 2016-06-25:
As the application itself says, "No devices connected". In other words, ckb is not detecting your device.
Which device do you have?

adamleblanc501, 2016-06-25:
Oh wow I didn't even realize that...I feel dumb lol. But I am using the K65.

tatokis, 2016-06-25:
Which K65

adamleblanc501, 2016-06-25:
Corsair K65 Lux RGB

tatokis, 2016-06-25:
Download and compile ckb from the newdev branch, as that contains support for the K65 Lux.
https://github.com/ccMSC/ckb/archive/newdev.zip

adamleblanc501, 2016-06-25:
Yes, that worked great. Thank you very much.