ccMSC/ckb #404: Firmware upgrade fails on M65 Pro

taisph, 2016-07-01:
When starting CKB it reports a firmware upgrade is available but any attempt to upgrade the firmware immediately fails.

Log: 
```
[E] fwupdate (firmware.c:78): Firmware file /tmp/ckb-3353-fwblob.bin doesn't match device (V: 1b1c P: 1b12)
```

tatokis, 2016-07-11:
That means that the firmware it is trying to install does not match your device, and any attempt to install it will lead to a bricked device.

taisph, 2016-07-11:
@tatokis Well, it is pretty obvious what it means. :smile: 

The problem is that CKB tool is suggesting the erroneous upgrade every time I start the tool. So in my opinion it should either be corrected to point to the correct firmware or, if that's not possible, stop suggesting the wrong firmware altogether.

tatokis, 2016-07-11:
I see now, I thought you were trying to upgrade it manually by specifying the bin file.
I'll have a look and see what can be done

tatokis, 2016-07-11:
Would you mind sending me an md5 or sha1 sum of the file in /tmp/?

taisph, 2016-07-11:
@tatokis Here's the sha1sum:
```
# sha1sum /tmp/ckb-4034-fwblob.bin
c44ef9f251f6e15f8d130798701d29039c857f4a  /tmp/ckb-4034-fwblob.bin
```