ccMSC/ckb #406: No devices connected with the latest version [K70 LUX RGB]

singleseeker, 2016-07-02:
Still shows me No devices connected with ckb beta-v0.26+d01.
I use  K70 LUX rgb.

colinazzam, 2016-07-06:
I am having the same issue. I'd be more than happy to provide any information or assistance that could help get this keyboard to work.

xiangyad, 2016-07-06:
Made it work by adding the hardware ID to some of the scripts here. More than happy to share the pkg.

xiangyad, 2016-07-06:
[ckb.zip](https://github.com/ccMSC/ckb/files/349212/ckb.zip)

ChiefPaleface, 2016-07-07:
I was having the same issues on Ubuntu for the k95 RGB. Not sure why but a reinstall did the trick.

tatokis, 2016-07-11:
The K70 Lux doesn't work because there wasn't support for it in the code.
It was added some time ago, and you need to build and install ckb from here https://github.com/ccMSC/ckb/archive/newdev.zip

stevenkendrick3, 2016-08-11:
Hey. I really hope that someone will respond to this. I recently bought a K70 LUX RGB keyboard. I am having the same problem where it says no devices connected. It seems like you solved the problem. I am just to stupid to understand how to fix it... I am by no means a programer so I was hoping that one of you could walk me through how to fix this? @tatokis  @singleseeker  @colinazzam  @xiangyad @ChiefPaleface

slugger7, 2016-10-08:
@xiangyad i would like to know what you did to fix the problem if that will be possible.

stripTM, 2017-02-11:
Hi, I have the K70 Lux (no RGB).

I have connected it to windows and I have updated the firmware.

I added in the kernel cmdline usbhid.quirks=0x1B1C: 0x1B09: 0x0x20000408 and restart

I installed the version of @tatokis ckb beta-v0.2.6 + d01

And I still can not get the message "No devices connected"

This is what I get if I execute **lsusb -d 1b1c: 1b36** show:
Bus 001 Device 003: Corsair ID 1b1c: 1b36

I would also like to know what is wrong, if you could at least orient me in the code that I have to find / add / change.

bigdan4712, 2017-03-25:
@tatokis YOU ARE THE MAN!

raviv, 2017-11-29:
@tatokis - Thanks !

tatokis, 2017-11-29:
For anyone possibly stumbling across this, update to https://github.com/mattanger/ckb-next

AntonisHopkins, 2018-02-22:
@tatokis  good jog man

riprip74, 2019-05-23:
Thx, installing ckb-next as mentionned above did the trick!

closed 2022-12-08 by singleseeker