ccMSC/ckb #408: K65 Lux Help!

SoupitySoup, 2016-07-08:
Hey, i just got the new k65 lux and tried to get this to customize it. I downloaded the mac versions and did all the instructions, i downloaded the pkg, got Xcode, and xt5, but it still says no devices connected. so then i just downloaded the main one and it also says no devices connected after trying to reboot my computer and replug my keyboard but still nothing. I'm using a mac. and the main one, when i open it says "Warning: driver version mismatch) beta-v0.2.6). Please upgrade cab-daemon. If the problem persists, try rebooting.

tatokis, 2016-07-11:
Install ckb from here https://github.com/ccMSC/ckb/archive/newdev.zip
There are instructions in the readme.

SoupitySoup, 2016-07-12:
Nope, still says the same thing. Followed the instructions exactly as it says.

mmsz111, 2017-01-28:
K70 Rapidfire
[image]

help...plz I tried all branches and rebooted.

mmsz111, 2017-01-28:
here... https://github.com/ccMSC/ckb/issues/570