ccMSC/ckb #412: the program can't find my keyborad, HELP!

TheDankMemeLord18, 2016-07-10:
The Program cant find my keyboard i keep unplugging it and plugging it back in but it still cant be found, can anyone help!?

zcfan, 2016-07-10:
I encounter the same problem before. Install the lasted version(0.2.6) and updated the framework solved my problem. k95 rgb, os x 10.11.5, hope it helps.

TheDankMemeLord18, 2016-07-10:
I got the K70 Lux RGB and im a bit of a newbie at this

piyatat, 2016-07-10:
I had the same problem as well.
I did a temp fix by changing product ID in the src file.
You can follow the instruction "Building from source" and change product ID in this file
"src/ckb-daemon/usb.h"

You can follow post in MEGATHREAD to find product ID of your kb.

TheDankMemeLord18, 2016-07-11:
ok thanks

horen20wash, 2016-07-11:
Hallo,

What would it take to add support fro K70 LUX RGB into the main build.

Product ID: 0x1b33
Version 1.05

Thanks

tatokis, 2016-07-11:
https://github.com/ccMSC/ckb/commit/e81f2c94ee8e46370f3eeb3a840a020df9935923

Support is already there, you have to build the newdev branch.
There are instructions in the readme.
If it still doesn't work, then there is an issue elsewhere.

horen20wash, 2016-07-11:
Thank you very much, will do.

horen20wash, 2016-07-11:
Used newdev to build, installed.
After reboot, the led over 1 , next to win lock is on and the keyboard non-functional. Unplug keyboard and plug again, keyboard works but cab shows no device connected.
Thanks

retrotransposon, 2016-07-16:
I built it from source manually and it works for my k65 lux as long as I dont reboot. Each reboot I have to trash the app and rebuild it in order to detect my keyboard. Any suggestions? its as if it reverts to a previous build or something.

horen20wash, 2016-07-16:
If you use the newdev branch and build and follow the instructions for copying the launch scripts, it all should work without any problems. You should be able to logout or reboot and the keyboard should work as you customized.

retrotransposon, 2016-07-17:
I used the newdev branch and it doesn't work on reboot. The easiest fixed is to quit the launch daemon and then it finds the keyboard.

retrotransposon, 2016-07-18:
Ok so after working with it some more I realized the problem is only due to switching the keyboard to bios mode. If I switch boot drives in system preference instead of holding option/alt at startup (which requires the keyboard to be in bios mode) then the scrip works fine on start up. If switched to bios mode and back then I need to restart the launch daemon.

Johnson-Z, 2016-07-24:
+retrotrandpodon
I have the problem on my k65 rapid-fire too, cab shows that no device connected.
Can you tell me how you solve the problem. :)